fritzing/fritzing-app #3504: Switch from qmake to cmake

KjellMorgenstern, 2019-07-09:
Pro
- Qt itself is switching from qmake to cmake
- cmake modules might provide better cross platform support than pkg-config
- better integration with conan

Contra
- Qt creator support for cmake not yet in par with qmake
- Quite some additional work
- Never change a working build system

cjmayo, 2019-07-21:
Another pro, if unbundling QuaZIP a FindQuaZip5.cmake is supplied as far back as the version supported in Ubuntu Bionic (18.04LTS). pkg-config support is in the repository but not released as yet.

GottfriedSp, 2019-09-03:
Contra:
•Support for qmake will continue unaffected
CMake for windows is realy bad.
CMake needs double source code, when compiling for 32-bit and 64-bit.
Much waste of ressources and time to change everything to CMake - new tests - new bugs - no improvement

CMake improvements is in development for Qt, this could be at least some years to be stable.
Qbs is better, it compiles only changed source files like MSVC. This speeds up re-compiling by 90% instead of makefiles.