fritzing/fritzing-app #3505: Upgrade to Qt >= 5.9, drop support for Qt < 5.9

KjellMorgenstern, 2019-07-09:


GottfriedSp, 2019-08-30:
Qt 5.9 does not work with Windows XP.

Latest supported Qt for Windows XP is 5.6.3

htho, 2019-09-17:
@GottfriedSp So what?

ccoenen, 2019-09-24:
You do not want to give people the impression that XP is any kind of viable option. So I would argue that missing XP support is a *good* thing.

htho, 2019-09-25:
@GottfriedSp On the other hand I am curious: Why did you mention this? Are there any reasons why XP support would be beneficial?

GottfriedSp, 2019-09-26:
I think, many computers on the world have windows XP.
For me, I have many old laptops and computers where only XP runs on them fine, and not debian or other linux runing on them, as they needs a lot of RAM and ressources and have some driver issues.
In my hobby workshop I use this old computers. They run fine and dirt and dust is regardless for them. And they have parallel ports and serial ports for special equipment I use. Parallel port adapters does not work.

But maybe I am wrong and one of the last who uses such old computers. I have fritzing 0.9.3 beta from official download page. It crashes sometimes and there is no parts editor with it.
When install XP again, you got all updates until its end of support.

ccoenen, 2019-09-27:
This will sound harsh, I'm sorry for that.

With that reasoning, we could also support any arbitrarily old operating system. I do not believe that this is a good idea. According to [this](https://netmarketshare.com/operating-system-market-share.aspx?options=%7B%22filter%22%3A%7B%22%24and%22%3A%5B%7B%22deviceType%22%3A%7B%22%24in%22%3A%5B%22Desktop%2Flaptop%22%5D%7D%7D%5D%7D%2C%22dateLabel%22%3A%22Custom%22%2C%22attributes%22%3A%22share%22%2C%22group%22%3A%22platformVersion%22%2C%22sort%22%3A%7B%22share%22%3A-1%7D%2C%22id%22%3A%22platformsDesktopVersions%22%2C%22dateInterval%22%3A%22Monthly%22%2C%22dateStart%22%3A%222019-05%22%2C%22dateEnd%22%3A%222019-08%22%2C%22segments%22%3A%22-1000%22%7D) XP is below 2% market share, but my argument is not with market share. As stated before, encouraging people to move on is a good thing. XP is a dead horse and supporting it will slow this project down.

htho, 2019-09-27:
@GottfriedSp I get your Point.
@ccoenen You are right - XP should not be supported by any modern software anymore.
Your second point is even more important:
A project that is low on ressouces should not be slowed down by supporting old Hard/Software.
(As a WebDev I know that supporting the old Internet Explorer is the worst - I would not put effort in an OSS Project that supports IE "just because").

RobertPHeller, 2019-09-28:
Well the version of QT5 available for CentOS 6 is 5.6.1.  Presently, 0.9.3 does in fact build and run on my CentOS 6 machine.

ccoenen, 2019-09-28:
You should be able to _run_ a compiled version just fine. Building would require that you obtain qt from a different source than the package manager, but should still compile fine.

RobertPHeller, 2019-09-29:
Except that any version *built* on a different Linux will be  built with a new glibc, and thus won't run. And I seriously doubt someone is going to build it with a *static* glibc (or really any other static libraries, including a static libgit2).  If I am going to run it on my CentOS 6 box I am going to have to build it myself from source.  Note: unlike WindowsXP,CentOS6 is not yet EOL.  It is still a currently maintained distro (for maybe a year or so anyway).  I would really rather not even try to built Qt5 from sources.

GottfriedSp, 2019-09-29:
I am not alone!
https://forum.arduino.cc/index.php?topic=636677.0

I tried to compile Qt 5.6.x on windows some years ago, but I gave up after 3 weeks with a fail. There are to much dependencies and it can't compile the whole framework. Only the basics compiled, but no unicode support, no 2D, no 3D, no WebKit, no database, no Qt Quick, ...
I think Qt is platform independed, but it is not."One Framework. One Codebase. Any platform." - Windows XP is also a platform.

GottfriedSp, 2019-09-29:
Windows 10 - 48%
Windows 7 - 33%
Mac OS X - 5%
Windows XP - 1.81%
Linux - 1.45%
Ubuntu - 0.46%
Debian - 0.00%
Windows XP have more market share than Linux!
But this are only machines connected to internet with a browser.

KjellMorgenstern, 2019-09-29:
To be clear: We don't have the resources to increase the number of supported systems. For me, that means moving on, adding more recent ones, and dropping the old ones. 
My friendly advice is: Keep the applications which are built for your favored OS version. Just continue using Fritzing 0.9.3, 0.9.4, or whatever the latest supported for your OS might be.

@RobertPHeller 0.9.4 still uses Qt 5.6; the switch is planned for 0.9.5.
This will affect CentOS 6 users, and also, for example, Ubuntu 16.04 users. With CI/CD system on Travis in place, it should however not be too difficult to derive a build that does not rely on the systems Qt installation or patch out the Qt 5.9 calls (it is not like the whole code is going to be re-written Qt 5.9-only in an instant)

@GottfriedSp: For Fritzing, the number of XP users might even be higher. However, this is not how the calculation works. I don't have a statistic, but I'd claim that the share of Linux developers who volunteer to contribute code to Fritzing is about 50%. And that is not all. The cost of supporting XP would be higher than the cost for all the other operating systems in total, both in money and developer time. For Microsofts extended support for XP, numbers like 250000$ were called, only targeting big companies, who are also required by Microsoft to create a migration plan and update it each quarter. This was in 2014, not sure if this program still exists. For consumers and small businesses, Microsoft is very clear about this for a long time: Use anything else, but not XP. Even if someone sent me an XP build of Fritzing, I would have no means to check if that is not just a bundle of viruses, so I could not publish it.

RobertPHeller, 2019-09-30:
Is there anything like a timeline (even approximate / hand waved) for when 0.9.5 is coming out? I am thinking I will be off CentOS 6 in about (hand waving) a year or so. Either I would have migrated by two C6 systems (Laptop and Desktop) to C7 or *replaced* the x86-flavored laptop with something ARM flavored (possibly a battery powered Raspberry Pi or a CubiTruck or a Banna Pi -- the CubiTruck and Banna Pi have SATA interfaces, which allows a *big* disk (>> 32G MicroSD)).  The various ARM Linux distros are much more "current" and at least with Raspbian I have figured out the stuff I need to "hack" (configuration wise) in the GUI Desktop to get something I can live with.  *NONE* of the "modern" desktop GUI systems are even remotely to my liking, at least not with major hackery in configuration land -- that is the *main* reason I have not upgraded my system yet.

WimberleyTech, 2019-09-30:
For what it is worth, I absolutely agree.
    
    On 9/29/2019 5:47 PM, Kjell wrote:
    
    
      
      To be clear: We don't have the resources to increase the number
        of supported systems. For me, that means moving on, adding more
        recent ones, and dropping the old ones.
        My friendly advice is: Keep the applications which are built for
        your favored OS version. Just continue using Fritzing 0.9.3,
        0.9.4, or whatever the latest supported for your OS might be.
      @RobertPHeller
        0.9.4 still uses Qt 5.6; the switch is planned for 0.9.5.
        This will affect CentOS 6 users, and also, for example, Ubuntu
        16.04 users. With CI/CD system on Travis in place, it should
        however not be too difficult to derive a build that does not
        rely on the systems Qt installation or patch out the Qt 5.9
        calls (it is not like the whole code is going to be re-written
        Qt 5.9-only in an instant)
      @GottfriedSp:
        For Fritzing, the number of XP users might even be higher.
        However, this is not how the calculation works. I don't have a
        statistic, but I'd claim that the share of Linux developers who
        volunteer to contribute code to Fritzing is about 50%. And that
        is not all. The cost of supporting XP would be higher than the
        cost for all the other operating systems in total, both in money
        and developer time. For Microsofts extended support for XP,
        numbers like 250000$ were called, only targeting big companies,
        who are also required by Microsoft to create a migration plan
        and update it each quarter. This was in 2014, not sure if this
        program still exists. For consumers and small businesses,
        Microsoft is very clear about this for a long time: Use anything
        else, but not XP. Even if someone sent me an XP build of
        Fritzing, I would have no means to check if that is not just a
        bundle of viruses, so I could not publish it.
      —
        thread.
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/fritzing/fritzing-app/issues/3505?email_source=notifications\u0026email_token=ALKDFNPEYRB3I5UBXIT2AC3QMEWB3A5CNFSM4H7DWVE2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD74AVAQ#issuecomment-536349314",
"url": "https://github.com/fritzing/fritzing-app/issues/3505?email_source=notifications\u0026email_token=ALKDFNPEYRB3I5UBXIT2AC3QMEWB3A5CNFSM4H7DWVE2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD74AVAQ#issuecomment-536349314",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

ccoenen, 2019-09-30:
@GottfriedSp you will run into this more and more. You are surely not alone. This is partly the point: Get off of XP. At least disconnect it from the internet. Your machine is a hazard to other people. There's (with very few high-profile exceptions) no security patching being done to it. Nobody should keep supporting XP. This is not an operating system for modern times. There are alternatives, even free ones, even on existing hardware. There is absolutely no point in catering to XP.

(and: [As i've said earlier](https://github.com/fritzing/fritzing-app/issues/3505#issuecomment-535852615): this argument **is not and was never** about market share).

KjellMorgenstern, 2019-10-10:
Done: https://github.com/fritzing/fritzing-app/pull/3541

closed 2019-10-10 by KjellMorgenstern