fritzing/fritzing-app #3510: Opening on random screens with i3wm

noobermin, 2019-07-10:
It's possible this might be a i3 issue but this doesn't happen with any other gui application I use currently. Fritzing randomly opens new windows (including opening new sketches) on seemingly random screens with multiple screens on the active workspace for that screen, not the currently active workspace/screen. Moreover, it grabs the mouse and moves it to that screen.

As for most tiling window managers i3 can use rules to place windows on preferred workspaces and screens but I have added none for Fritzing.

KjellMorgenstern, 2019-07-10:
Which version of Fritzing did you use? Can you reproduce the problem with the latest development build? (available via github releases)

KjellMorgenstern, 2019-10-02:
This issue was closed due to missing information.
If you think this issue is still important, please provide additional information.
  - Which version of Fritzing have you been using?
  - On which operating system?
  - Please add screenshots with error messages or rendering issues. Mark the problem with an image editor.

closed 2019-10-02 by KjellMorgenstern