fritzing/fritzing-app #3522: Subscribing to Fritzing forum is not possible

KjellMorgenstern, 2019-08-23:
## Current Behaviour
Fritzing forum signup is not possible

**Steps to reproduce:**
Signup with a new email.
Fritzing forum will tell you to activate via the confirmation email.
- Email does never arrive
- When trying to log in, error message "This account is currently inactive" is shown.

## Expected Behaviour
Email should be sent so the account can be activated.

KjellMorgenstern, 2019-08-24:
Looking into the database shows that on 2019-07-17, about 4242 accounts had been registered to fritzing.org . To me this looks like someone tricked the signup captacha, and created a kind of DoS for the signup.  For more recent subscriptions, no mail has been sent (yet).

KjellMorgenstern, 2019-09-19:
There was an error in the DKIM settings, which would explain emails delivered (or better: accepted by other servers) slow or not at all. This is corrected now, fingers crossed it was the only issue.

KjellMorgenstern, 2019-09-27:
Despite the DKIM fix and successful Mandrill check, emails are still not delivered five days later. Checking again...

KjellMorgenstern, 2019-10-01:
After learning a lot about the history of Mandrill (and Mailchimp), I think the root cause was that Mailchimp had changed their tariffs in May/June. 
As Mailchimp is overkill for managing a forum, I switched to Mailgun. Main reason to use mailgun: They had good documentation for using a similar smpt configuration, so no code changes on the fritzing.org server needed. 

Signup and Password changes should work again.

closed 2019-10-01 by KjellMorgenstern

KjellMorgenstern, 2019-10-02:
Just for the records: I have extracted all the accounts that have been created since the email service stopped, and are know sending those mails via a script. There is some careful delay, so it will take some days until all mails are sent (many thousand registrations). Especially if you have a Yahoo account, please check your spamfilter, I see quite some bounces from there.

imWACC0, 2024-09-02:
Ok, it's a few years later, did this ever get fixed?

If yes, can I get a link to join, I can't seem to find it.