fritzing/fritzing-app #3523: Building with windows does not work (wiki)

GottfriedSp, 2019-08-30:
>c:\Qt\5.6.3\msvc2015\bin\qmake.exe -t vcapp phoenix.pro
Info: creating cache file E:\fritzing-app\.qmake.cache
Project MESSAGE: target arch: x86
Project MESSAGE: Fritzing requires libgit2
Project MESSAGE: Build it from the repo at https://github.com/libgit2
Project MESSAGE: See https://github.com/fritzing/fritzing-app/wiki for details.
Project ERROR: libgit2 include path not found in E:/fritzing-app/../libgit2/include

Windows 7

vanepp, 2019-09-09:
You need to install both boost and libgit2 before trying the build. From the error message there is something wrong in your libgit2 install (it isn't particularly easy on Windows). 

Peter

KjellMorgenstern, 2019-09-10:
@GottfriedSp  I updated the wiki page you referred to. Compiling boost is not required, as we currently only use boost header libraries. So, what the error message is saying: Just extract boost next to your fritzing source directory, and it should be fine. In theory, it should also find a boost system installation, but this is currently untested.

GottfriedSp, 2019-09-10:
This is great!
Ok, i try it again, thank you!

vanepp, 2019-09-10:
Sorry, because I didn't see the boost error message I assumed you had boost. I didn't know we are only using the headers! I usually copy in the precompiled Windows binaries available on the boost site to fritzing-app/src/lib/boost1.xxx . That is much easier than compiling boost. 

Peter

KjellMorgenstern, 2019-09-19:
Did it work?

closed 2019-09-20 by KjellMorgenstern