fritzing/fritzing-app #3525: View from Below for Breadboard

datacute, 2019-09-14:
## Problem
I use the breadboard view to design through-hole modules on prototyping boards, and I would like to be able to design what goes on top of the board, and what goes underneath the board.

Currently, I've modified the prototyping board's parts file to be flippable, and I end up with three copies of everything - one with everything connected, one with just what is going to be on the top, and one flipped, with just the components and wires that are on the bottom.

This messes up the Schematic and PCB views.

## Proposed Solution
I'd like to be able to have the PCB window's capability to view from above or from below, and to either be able to select whether to show top and bottom layers, with the ability to specify which layer components are on, along with components being able to be shown on both layers, or possibly just reversing the z-order of displayed components (including wires and boards), might be a workable starting point.

KjellMorgenstern, 2019-09-14:
Like, a using a double sided perfboard?

datacute, 2019-09-14:
Yes

wraybowling, 2019-09-23:
This is an interesting idea but solely for perfboards and stripboards. While it would be convenient for documenting circuits built like that in an accurate way, this feature would potentially wreak some havoc on how reliably breadboard layouts can be printed or converted to SVGs for sharing and education.

KjellMorgenstern, 2019-09-24:
@wraybowling Do you have a specific code smell or experience with Fritzing in mind? Without this, your concern sounds a bit like the generic „new stuff might break existing“. 
Having automated tests for SVG export would certainly be a plus, of course.

wraybowling, 2019-09-24:
@KjellMorgenstern My response was based solely on the fact that screenshots are not two-sided. I'm a user experience designer, not really a unit test writer, so I can speak only to the experience of users wanting to export their breadboard layouts and put them online or in books. We'd need to render the back sides of those exports, perhaps side-by-side on a single image?

ccoenen, 2019-09-24:
I strongly feel that this is a rare edge case, which should be prioritized _way_ down, bordering on "won't be done". On top of that, this might confuse newcomers and makes export harder (as mentioned by @wraybowling).

datacute, 2019-09-29:
An alternative solution might be easier to implement with less changes:
1. In the preferences for Breadboard View add a checkbox to allow Z-ordering of wires components and breadboards together instead of always having wires above components above boards.
2. When a breadboard or stripboard or protoboard shield is flipped horizontally or vertically, also flip all components attached to the board, and if the new preference is turned on, also reverse the z-order of everything attached to that board.
Then a board can be positioned in the z-order with components above and below it, and by flipping a board the view would effectively change to the opposite side.

KjellMorgenstern, 2019-09-30:
The label "enhancement" unfortunately means that this issue gets quite a low priority, as there are many open bugs, even critical ones. From the feature implementation side I wouldn't even estimate it as very difficult. The UX should behave very similar to what we already have for the PCB view, and the current view should be the default. I'd restrict the Z-Level for breadboards however, the feature makes more sense on stripboards and the like ;-)

Djeez2, 2020-02-25:
Having a bottom view of the stripboard is my #1 wish for a long time. My use case:

- I design the top adding components and cut strips where necessary.
- Then, I take the real physical stripboard to start "production". However, there are 2 problems:

1. The component block the view on the stripboard (e.g. typically underneath an IC you would cut the strips)
2. On the actual board the strips are on the bottom side (opposite the components) so I have to mirror the image in my head.

A bottom view that is identical to the top view but with all components removed and mirrored would fit my needs.

Thank you for considering this enhancement.

KjellMorgenstern, 2020-02-25:
1. How about disabling the parts layer to see the cuts?
In the Breadboard view, menu -> view ->  [ ] parts layer

2. That makes perfect sense.

Djeez2, 2020-02-26:
Thanks voor the view tip, I didn't know that one and it certainly will help.
I'm glad you continue on the development of Fritzing, there is no other software that suits my needs that well.

HansElectrumpet, 2022-09-05:
For educational purposes I would like the flip option. My use case is: transferring a schematic from breadboard to perfboard. Then, being able to flip it upside down, maybe with a semi transparent perfboard, students can easily watch the backside of the board. Many mistakes are made confusing left and right. Even more so for people who have problems keeping those apart which is some sort of brain condition it seems...

KjellMorgenstern, 2022-09-06:
@HansElectrumpet Have you considered using the PCB view for a perfboard? In theory, it should be possible to add a PCB in the layout of a perfboard. 

Disadvantage: 
Components in PCB view look more abstract, mostly footprints and some guides in silkscreen.

Advantage: 
Both sides can be used already.
Ratsnest lines (dotted lines) will show in case the net deviates between PCB, schematic and breadboard.