fritzing/fritzing-app #3527: New Parts Editor: Schematic view breaking my custom svg

Constikri, 2019-09-16:
## Current Behaviour
When i load in my svg file, that i made with inkscape, some strange rectangles appear, that are partly covering the schematic.

[image]
[image]
[image]

**Build:**
Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.6.1]

**Operating System:**
Windows 10

**Steps to reproduce:**
- receive my svg file somehow
- load it into any part in the parts editor

## Expected Behaviour
My svg looks the way its supposed to, like every other svg, that is loaded into the parts editor.

maclof, 2019-09-18:
I've been making a lot of parts these past few days, first I started out having similar issues like this, however through lots of testing I managed to get my process down to a few steps. I'll point out that I've had no luck with Inkscape, the SVGs it generates are very messy in comparison to existing Fritzing svgs I've looked at and I was never able to get the parts looking right, these points below may still apply for you though, I now use Adobe Illustrator and it generates near perfect SVGs compatible with Fritzing.

1. Generate the SVG using mm or in as the dimension, the svg must not be generated as responsive and should include width and height attributes in the main <svg> opening tag.
2. Change inline styles to attributes, for example Illustrator generates some objects and puts inline styles:
`style="fill: none;stroke: #fcbe0e;stroke-width: 1.440000057220459px"`
I change these to attributes, such as:
`fill="none" stroke-width="2"`
This fixes the strange black rectangle behaviour your experiencing.
3. All pins must be assigned to part of the SVG, if you do not do this, you'll get a big red square appear when trying to use the part in svg/pcb view making the part unusable. This may sound like a weird one but I started out testing my parts to see how they looked before I'd assigned all pins, took me ages to figure out why Fritzing was placing a random red rectangle in the middle of the part.

I've attached an example of an SVG that if you load it into Fritzing (pcb) will show the black square and a second SVG with the with the black square issue fixed. Compare the SVGs and you'll see what I did with the inline styles.

[image]

[image]

[black-square-example.zip](https://github.com/fritzing/fritzing-app/files/3627709/black-square-example.zip)

KjellMorgenstern, 2019-09-18:
Confirmed that the issue still exists in the development version CD-346 ( 32ab1e86 ).

vanepp, 2019-10-10:
@Constikri  if you would upload the svg to either the fritzing forums (where you will likely need to rename the svg to a .fzp file as the forums often have trouble rendering svgs) I'll have a look at it and see if I can tell you what is wrong with it. None of the editors are a good fit for Fritzing, I have a python script that corrects a number of the things Inkscape does that break fritzing (there is also a post in the forums on settings for Inkscape that work the best with Inkscape. There is a current thread in the forums from an Illustrator user that is having problems. The python script changes style commands to in line xml and removes the px from font sizes (which break the parts editor as it sets the font size to 0 on a re edit) among other things. The script is available from 

https://github.com/vanepp/FritzingCheckPart

@maclof  your suggestions on how to use illustrator for fritzing would be much appreciated in the forums (I use strictly Inkscape and am thus of not much help other than to tell them why the xml is not to Fritzing's liking). 

Peter

wraybowling, 2021-01-19:
I think we can close this issue. There's been no updates. Somewhat related however, and perhaps useful to a future lurking reader, I made this set of Inkscape templates for making new Fritzing parts. https://github.com/MicroWrave/fritzing-new-part-kit A lot of the issues folks have with SVG incompatibilities have to do with Fritzing expecting groups that have very specific IDs and being nested in a very specific way. The best way I know how to solve for any confusion is with this carefully tailored template that has all of the groups already created in the right structure. I don't have access to a new version of Illustrator, but I can see if there's a way to do the same with an older version.

KjellMorgenstern, 2021-07-13:
New protoype of the parts editor (not yet published) points out the issues with the SVG (duplicate definition of fill, stroke and stroke-width).

closed 2021-07-13 by KjellMorgenstern