fritzing/fritzing-app #3528: [LEAP 15.0][Linux] fritzing can't work than once

truesoundlord, 2019-09-17:
I tried fritzing had time to edit two files and then no way to launch the application again.

[image]
[image]
[image]

I tried to wipe out all related to fritzing and even from clean install the application can not find the parts I just used to create my two fzz files.

KjellMorgenstern, 2019-09-17:
I guess you were using Fritzing Version 0.9.3beta? If yes, would you like to try the development build? It can downloaded here on github.

truesoundlord, 2019-09-17:
Sorry I did not dig very far in depth... here this is the beta version, the github version crashes KDE forcing me to kill the "startkde" process to get back to a working environment :{

Workaround: write a bash shell that removes the parts.db file that has a zero byte size.
Once done, the app is able to run each time launched :}

KjellMorgenstern, 2019-09-18:
Ah, LEAP 15.0 is a Suse Linux. The development version are currently only build for recent Fedora and Ubuntu versions, sorry. Trying to run it on Suse usually should lead to an error because of a missing library, not crash kde.

We don't know why sometimes the parts.db file is corrupted on the old 0.9.3beta. The development build already has a fix to automatically recover from zero size parts.db files.

KjellMorgenstern, 2019-09-20:
Closing this, as we don't plan to actively provide binaries for all possible Linux distributions. We would however support or integrate AppImages or similar if someone would create such a PR.

closed 2019-09-20 by KjellMorgenstern