fritzing/fritzing-app #3530: Invalid text color for Notes under macOS

andrivet, 2019-09-21:
## Current Behaviour
Under macOS (Mojave, 10.14.6), the color of the text in Notes is white (was black in previous version of Fritzing) and thus hard to read. When you select the text, the background (of the text only) becomes black.

Something that may be important: I am using Dark Mode under macOS.

[image]

**Build:**
Version 0.9.4 (b872b5cd 2019-07-05) Cocoa [Qt 5.12.2]

**Operating System:**
macOS (Mojave, 10.14.6)

**Steps to reproduce:**
- Create a note
- The text color is white
- Select the text
- The background of the text becomes black

## Expected Behaviour
- The text has to be dark with a transparent background

2E0PGS, 2020-01-25:
I have a similair issue under Linux. I suspect it's trying to follow my system themeing.

[image]

KjellMorgenstern, 2020-01-27:
@2E0PGS Which theme are you using? Or is it a custom one?

KjellMorgenstern, 2020-01-31:
Related to #3336

commit 28bf1839 (2020-02-05): Override system color for notes. Fixes #3530
closed 2020-02-05 by KjellMorgenstern