fritzing/fritzing-app #3532: Build error under CentOS 6

RobertPHeller, 2019-09-28:
## Current Behaviour
g++ -c -pipe -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -m64 -mtune=generic -O2 -std=gnu++0x -Wall -W -D_REENTRANT -fPIC -DLINUX_64 -DDATADIR=\"/usr/share\" -DPKGDATADIR=\"/usr/share/fritzing\" -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_NETWORK_LIB -DQT_SERIALPORT_LIB -DQT_SQL_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -I/home/heller/libgit2/include -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtPrintSupport -isystem /usr/include/qt5/QtSvg -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtConcurrent -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtSerialPort -isystem /usr/include/qt5/QtSql -isystem /usr/include/qt5/QtXml -isystem /usr/include/qt5/QtCore -Irelease -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o release/connectoritem.o src/connectors/connectoritem.cpp
src/connectors/connectoritem.cpp: In static member function ‘static void ConnectorItem::collectParts(QList<ConnectorItem*>&, QList<ConnectorItem*>&, bool, ViewLayer::ViewLayerPlacement)’:
src/connectors/connectoritem.cpp:1402: error: expected primary-expression before ‘[’ token
src/connectors/connectoritem.cpp:1402: error: expected primary-expression before ‘[’ token
src/connectors/connectoritem.cpp:1402: error: ‘clang’ has not been declared
src/connectors/connectoritem.cpp: In member function ‘bool ConnectorItem::legMousePressEvent(QGraphicsSceneMouseEvent*)’:
src/connectors/connectoritem.cpp:2461: error: expected primary-expression before ‘[’ token
src/connectors/connectoritem.cpp:2461: error: expected primary-expression before ‘[’ token
src/connectors/connectoritem.cpp:2461: error: ‘clang’ has not been declared
make[1]: *** [release/connectoritem.o] Error 1
make[1]: Leaving directory `/home/heller/fritzing-app-new'
make: *** [release] Error 2

**Build:**
Cloned from github commit 9ad159e60c52b3e4eceee283789896d83ebd6c8b

**Operating System:**
CentOS 6:
QT5 5.6.1, gcc-4.4.7, glibc 2.12
static build of libgit2 0.28.1

**Steps to reproduce:**
qmake-qt5
make

## Expected Behaviour
Successful build

KjellMorgenstern, 2019-09-29:
Change introduced by me with commit fb6f0d . Seems the way I silenced the warning is not downward compatbile with gcc4.

RobertPHeller, 2019-09-30:
Does this mean that Fritzing > 0.9.3 is no longer compilable under CentOS 6 (or any other Linux with gcc4)?  Or will the fix you made commit  fb6f0d will be revisited?

KjellMorgenstern, 2019-09-30:
I wouldn't mind a workaround for this particular issue. See also https://dzone.com/articles/implicit-fallthrough-in-gcc-7 about it (not taking clang or MS Visual C++ into account). Not sure however if that would solve the general issue.
There is no explicit support for gcc 4 anymore. We plan to allow C++14 from 0.9.5 and later on, see #3483

RobertPHeller, 2019-09-30:
Attahed is a patchfile that fixes things to compile under CentOS 6.  I also built a static libgit2 library, since the installed (repo supplied) version of libgit2 is too old.

[CentOS6-Fritzing-0.9.4.patch.txt](https://github.com/fritzing/fritzing-app/files/3671107/CentOS6-Fritzing-0.9.4.patch.txt)

KjellMorgenstern, 2019-10-10:
Qt 5.9 was configured on the develop branch, and C++14 support is in the pipeline.
Older build targets have been removed from the develop branch. 
I started a legacy support branch, that could receive packports. The gcc4 fix was added there as well. Note: The CentOS6 patch also disabled dynamic linking of libgit2. I added an according variable to phoenix.pro file.
https://github.com/fritzing/fritzing-app/pull/3534

closed 2019-10-10 by KjellMorgenstern

RobertPHeller, 2019-10-10:
OK, everything compiles under CentOS 6 using the legacy build branch, except it fails to link.  Missing system libraries:  -lssh2 -lhttp_parser are needed by libgit2.a

KjellMorgenstern, 2019-10-10:
Blindly including those system libraries breaks on ubuntu and fedora.

RobertPHeller, 2019-10-10:
Hmmm.  I guess some tests are needed...  The libgit2 CMakeLists file checks for them.  Maybe those tests could be adapted for qmake?  The libgit2 *shared* library has them as run time dependencies (according to ldd).  The static library does not pull them, so they need to be on the link comand for Fritzing.  Or maybe something different for Ubuntu and Fedora.  libgit2 references symbols in those libraries (or some library with those symbols).  ssh2 abd http_parser are deps if libgit2, so since libgit2 is a dependency of Fritzing, these two packages are also dependencies of Fritzing...

commit 22dc4356 (2019-10-10): Use a better supported fallthrough statement to fix #3532
commit 82f069fc (2019-10-10): Use a better supported fallthrough statement to fix #3532