fritzing/fritzing-app #3537: Gerber rendering moves and skips nodes in board outline

deshipu, 2019-10-05:
## Current Behaviour
This is an old bug, and it is probably already reported several dozen times here, but most of bug reports here are done by inexperienced users who don't really provide much information, so I decided to report this again, but this time provide as much information as humanly possible.

The issue is that when we have any moderately complicated custom board outline, the gerbers that are generated tend to randomly move or skip nodes in the paths in the SVG image used as the board outline. For instance, an outline like this (for a keyboard):
[image]
Results in a gerber outline that looks like this:
[image]

Interestingly enough, this can be sometimes worked around by manipulating the order and type of the nodes, reversing the order of nodes in the path, flipping the SVG image and then flipping it in Fritzing to undo it, etc. — however, with large enough number of nodes, this can take several hours to get right, and eventually becomes completely impossible.

I am attaching example files (the SVG image, the Fritzing file that uses it, and the resulting Gerber outline) here: 
[x.zip](https://github.com/fritzing/fritzing-app/files/3693626/x.zip)

I would love to help fixing this, if possible, because that bug (or, more likely, a set of bugs) has already caused me to waste hours of time and some money (remember to always verify your gerbers before ordering).

**Build:**
Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.9.1]

**Operating System:**
<Linux>

**Steps to reproduce:**
- Create a new project in Fritzing.
- Set the PCB to use a custom board outline from the ``keyboard-spaced_pcb.svg`` file.
- Export the design as a Gerber file.
- Use ``gerbv`` or any other gerber viewer to see the ``keyboard-spaced_countou.gm1`` file.
- See it's different from the expected outline.

## Expected Behaviour
The expected behavior is to have the gerber file show the same outline as the SVG file says (and has Fritzing displays in all the previews).

deshipu, 2019-10-05:
The bug is also present in the current master, 0.9.4 (b872b5cd 2019-07-05) 64 [Qt 5.9.5]

deshipu, 2019-10-05:
For a comparison, an SVG image of the same outline that generates correct gerber:

[keyboard-spaced_pcb.svg.zip](https://github.com/fritzing/fritzing-app/files/3693686/keyboard-spaced_pcb.svg.zip)

vanepp, 2019-10-10:
There was a major re write of the gerber code (some 6000 commits) that was backed out about 2 years ago due to problems. It is possible a fix may be somewhere in that code base which should still be in a branch in the source repository. Otherwise the source is available and we know how to build development systems again (for several years I couldn't get the Windows version to run, and I have no Macs), there are plans for a new release so a fix could actually get released (the gerber stuff was backed out in the hope we could do a release at that point but it never came to pass.) if someone is willing to do the work. 

Peter

deshipu, 2019-10-10:
I'm not very good with C++ or familiar with Fritzing, but I looked at the code for the gerber generator and I think I have some general idea about how it is done — and it's not pretty. I'm not sure I will be able to find the bugs responsible for this (there are likely more than one), but I already see some ways in which the generator could be improved (for example, don't generate movement commands for every single pixel along a straight line...). I will try to find that branch and see if I can get that to build.

vanepp, 2019-10-11:
"I'm not very good with C++ or familiar with Fritzing," Me either, (although I'm fairly familiar with Fritzing), but I don't let that stop me. Kjell had to correct my C++ style which caught what was probably a code error (even though it worked) in the declarations for my fixes for the seg fault when loading a bad part and the fix so Parts Editor will correctly delete the parts files, and the two fixes took me months of tracing with gdb to figure out how the code worked and how a fix could work. They are both crude, but fix two nasty bugs (and are in the 0.9.4 pre release which I am running). As far as I can tell there is no one (at least who will answer questions or provide advise) who knows the code active in the project any more, so if we want it to survive we have to step up and do what we can. Kjell and his team appear to have a good handle on recreating the build environment (I tried for 2+ years to get a build environment to work without much success), so if we can attract people willing to help Fritzing can survive.

KjellMorgenstern, 2019-10-12:
See #3375 for the new Gerber export

KjellMorgenstern, 2019-10-30:
Did you do anything in that direction yet? In general the new gerber export tried to do clean vector calculations (but obviously there where problems), and much less back and forth between vector and raster graphics. It's a good approach, but basically got some of the metadata wrong.

deshipu, 2019-10-30:
So far I only looked at the existing code, and got some basic understanding. I was on vacation this month, and didn't really have time to look at the new gerber export code yet. I'm not going to have much time to devote to this, but hopefully if I chip at it bit by bit, I might eventually get somewhere. But if someone else has the time and motivation to work on this, please don't look at me.

vanepp, 2019-10-31:
Unfortunately people with the skill and willingness to help are in short supply, thus at least in my view, you are likely our best hope unless someone else chooses to step up (and in a couple of years of pushing to restart development, I haven't seen much of anyone stepping up, so please do as much as you can!) I have a lot of work on parts to do, and know very little about gerber processing, but I'm stil of the view that slow as I am some progress is better than none. Two of my most annoying bugs are fixed in 0.9.4 even if it took almost 2 years to get them there. 

Peter

KjellMorgenstern, 2019-10-31:
Ok. I am working on the pcb view / gerber export and just wanted to align in case someone else also does.

vanepp, 2019-10-31:
What good timing, I just came across a path bug in the gerber export code. This is a clone of the Arduino Uno part to make a Uno Wifi rev2 part for someone. The first fzpz file has all the groups and translates in pcb removed, but the path outline for the board screws up (only in gerber output, Fritzing renders it correctly in pcb view.) The second fzp has a transform in silkscreen which makes the path work (it is a little lopsided but much better than the broken one). 

[Arduino-Uno-WIFI-REV2-broken.fzpz.zip](https://github.com/fritzing/fritzing-app/files/3796094/Arduino-Uno-WIFI-REV2-broken.fzpz.zip)

[Arduino-Uno-WIFI-REV2.fzpz.zip](https://github.com/fritzing/fritzing-app/files/3796097/Arduino-Uno-WIFI-REV2.fzpz.zip)

(remove the trailing .zip to get the fzpz files)

Peter

KjellMorgenstern, 2019-11-02:
Intermediate result, maybe it already helps.
Fritzing seems to have a problem with the relative lines after the svg m oveto command in combination with the V ertical line

Only difference is in the second line 
The code 
```
m 182.808,0.288 4.312,4.304 
v 32.276 
l 7.183,7.172 
v 92.526 
l -7.183,7.172 
v 4.351 
c 0.159,1.396 -0.841,2.66 -2.233,2.824 -0.189,0.021 -0.396,0.021 -0.593,0 
H 3.25 c -1.397,0.164 -2.663,-0.836 -2.827,-2.229 -0.023,-0.197 -0.023,-0.396 0,-0.596 
V 3.112 
C 0.26,1.715 1.261,0.45 2.658,0.288 2.855,0.265 3.054,0.265 3.25,0.288 
h 179.558
```
see https://codepen.io/morgens/pen/VwwyywV
(codepen doesn't render the bottom line, but is correct in other viewers)

is more or less transformed into

```
m 182.808,0.288 4.312,4.304 
L 4.3 32.276 
l 7.183,7.172 
v 92.526 
l -7.183,7.172 
v 4.351 
c 0.159,1.396 -0.841,2.66 -2.233,2.824 -0.189,0.021 -0.396,0.021 -0.593,0 
H 3.25 c -1.397,0.164 -2.663,-0.836 -2.827,-2.229 -0.023,-0.197 -0.023,-0.396 0,-0.596 
V 3.112 
C 0.26,1.715 1.261,0.45 2.658,0.288 2.855,0.265 3.054,0.265 3.25,0.288 
h 179.558
```
https://codepen.io/morgens/pen/PooEEoM
Looks exactly like the error in Fritzing

KjellMorgenstern, 2019-11-02:
The example from @deshipu now produces this .gm1 file:

[image]

KjellMorgenstern, 2019-11-02:
Ok... not yet. Seems like we have one more problem here. The broken outline from the Arduino Uno WIFI REV2 broken examples seems to be fixed, but the gm1 contour file of the keyboard is still broken.

[image]

KjellMorgenstern, 2019-11-04:
5 killed bugs later...
[image]

closed 2019-11-09 by KjellMorgenstern

KjellMorgenstern, 2019-11-09:
@deshipu Thanks a lot for the summary, that helped fixing quite a bunch of bugs. Going through the issue backlog and bringing up relevant ones is incredible valuable, especially since I only joined the project this year, and know nothing about the quirks of older Fritzing versions. More than 50% of the issues seem to be already fixed, or maybe users gave incomplete information of how to reproduce.

deshipu, 2019-11-10:
Sorry for a late reply, I was travelling. Wow, this is really great! You can't even imagine how much time and frustration this is going to save me. Thank you a lot!

vanepp, 2019-11-10:
I agree, this is great news! I have a long list of current bugs that I can reproduce that I haven' t entered yet, figuring I'd give you time to clean out as much of the back log as possible before adding to the work load. Maybe it is time to clear most of the older issues and start reporting new ones?

Peter

KjellMorgenstern, 2019-11-10:
@vanepp Don't hold back. Issues get more weight if reported by multiple people. Or if they can be reproduced on the most recent version of Fritzing. It is very important to link duplicate issues Otherwise, one issue gets fixed, and years later, someone wastes time when trying to reproduce the duplicate issue which was left open.

deshipu, 2019-11-10:
I have built the latest master (Version 0.9.4 (CD-415 2019-10-16) 64 [Qt 5.9.5]) and tried with that keyboard PCB, but after editing it in Inkscape, and it seems like there are still some problems:
[image]

The SVG file I tried: 
[keyboard-v2b_pcb.svg.zip](https://github.com/fritzing/fritzing-app/files/3828918/keyboard-v2b_pcb.svg.zip)

KjellMorgenstern, 2019-11-10:
The issue is fixed for 0.9.5, which is not yet on master, but on the develop branch

deshipu, 2019-11-10:
I see, sorry. I found a problem with the new regexp, though, left you a note in the commit comments.