fritzing/fritzing-app #3543: Additional libraries are required to build on Linux

cmbahadir, 2019-10-12:
## Current Behaviour
I have tried to build Fritzing on my Linux Mint PC and it ended up with "undefined reference" errors due to missing linked libraries -lcurl - lhttp_parser -lssh2

```
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `git_curl_stream_global_init':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:44: undefined reference to `curl_global_init'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:50: undefined reference to `curl_global_cleanup'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `apply_proxy_creds':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:75: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:77: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `curls_connect':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:128: undefined reference to `curl_easy_perform'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:130: undefined reference to `curl_easy_getinfo'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:146: undefined reference to `curl_easy_getinfo'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `curls_certificate':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:172: undefined reference to `curl_easy_getinfo'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `curls_set_proxy':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:215: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:218: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `curls_write':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:261: undefined reference to `curl_easy_send'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `curls_read':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:283: undefined reference to `curl_easy_recv'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `curls_close':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:299: undefined reference to `curl_easy_cleanup'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o): In function `git_curl_stream_new':
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:326: undefined reference to `curl_easy_init'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:338: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:339: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:340: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:341: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:342: undefined reference to `curl_easy_setopt'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(curl.c.o):/home/cmb/Workspace/Fritzing/libgit2/src/streams/curl.c:343: more undefined references to `curl_easy_setopt' follow
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(http.c.o): In function `clear_parser_state':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/http.c:504: undefined reference to `http_parser_init'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(http.c.o): In function `http_connect':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/http.c:596: undefined reference to `http_should_keep_alive'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(http.c.o): In function `http_stream_read':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/http.c:742: undefined reference to `http_parser_execute'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/http.c:768: undefined reference to `http_errno_description'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(ssh.c.o): In function `ssh_error':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:55: undefined reference to `libssh2_session_last_error'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(ssh.c.o): In function `send_command':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:115: undefined reference to `libssh2_channel_process_startup'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(ssh.c.o): In function `ssh_stream_read':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:142: undefined reference to `libssh2_channel_read_ex'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:153: undefined reference to `libssh2_channel_read_ex'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(ssh.c.o): In function `ssh_stream_write':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:181: undefined reference to `libssh2_channel_write_ex'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(ssh.c.o): In function `ssh_stream_free':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:209: undefined reference to `libssh2_channel_close'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:210: undefined reference to `libssh2_channel_free'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:215: undefined reference to `libssh2_session_disconnect_ex'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:216: undefined reference to `libssh2_session_free'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(ssh.c.o): In function `ssh_agent_auth':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:297: undefined reference to `libssh2_agent_init'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:302: undefined reference to `libssh2_agent_connect'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:307: undefined reference to `libssh2_agent_list_identities'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:313: undefined reference to `libssh2_agent_get_identity'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:327: undefined reference to `libssh2_agent_userauth'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:340: undefined reference to `libssh2_agent_disconnect'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:341: undefined reference to `libssh2_agent_free'
/home/cmb/Workspace/Fritzing/fritzing-app/../libgit2/build/libgit2.a(ssh.c.o): In function `_git_ssh_authenticate_session':
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:357: undefined reference to `libssh2_userauth_password_ex'
/home/cmb/Workspace/Fritzing/libgit2/src/transports/ssh.c:364: undefined reference to `libssh2_userauth_publickey_fromfile_ex'
```

**Build:**
Latest develop branch.

**Operating System:**
Linux Mint

**Steps to reproduce:**
- qmake --> successful
- make --> errored

ı am able to create a PR to solve it by adding the required libs on phoenix.pro.

cmbahadir, 2019-10-12:
Ok it seems like that the cause is my local config since the required libraries are installed under "/usr/lib/x86_64-linux-gnu" and they are not linked automatically. I realized it after failing travis build of my PR and success on previous ones :)

closed 2019-10-12 by cmbahadir

calocedrus, 2020-03-13:
@cmbahadir I have the exact same issue with: ubuntu 18.04, libgit2 (checkout v0.28.4, built locally), master fritzing-app. But I can't solve it: could you please explain how you add the missing linked libraries to phoenix.pro, or how else one can solve this?

cmbahadir, 2020-03-28:
Hi @calocedrus ,

I have just seen your message, actually i had openned an PR for this but closed afterwards, since it was like my local config problem.

Please take a look at it #3544 .

It is good to know that i am not the one who faced this issue :+1:

jcalcote, 2020-11-30:
The trick (it seems) is to remove libssh2(-1) from your system (along with any packages that depend upon it, of course). Then rebuild libgit2. It will use an internal ssh library.