fritzing/fritzing-app #3546: Building on Fedora doesn't link to pcre

tinpotnick, 2019-10-21:
## Current Behaviour
When you build on fedora the linker complains about the lack of pcre_free. 
<please attach screenshot if possible>
**Build:**
Git clone as of today. 
Version 0.9.5 
(dCD-380-63-gca1a1bfd 2019-10-20) 64 [Qt 5.12.5]

**Operating System:**
Linux - Fedora 30

**Steps to reproduce:**
- Build using normal instructions
- Get linker error

## Expected Behaviour
It to build.

I added '-lpcre' to the LIBS line in Makefile.Release and then it builds ok. I am not sure this is the correct place to put it as I think this file is auto generated?. My complete LIBS line now looks like:

LIBS          = $(SUBLIBS) -lz /home/nick/workspace/fritzing-app/../libgit2/build/libgit2.a -lssl -lcrypto /usr/lib64/libQt5PrintSupport.so /usr/lib64/libQt5Svg.so /usr/lib64/libQt5Widgets.so /usr/lib64/libQt5Gui.so /usr/lib64/libQt5Concurrent.so /usr/lib64/libQt5Network.so /usr/lib64/libQt5SerialPort.so /usr/lib64/libQt5Sql.so /usr/lib64/libQt5Xml.so /usr/lib64/libQt5Core.so /usr/lib64/libGL.so -lpthread -lpcre

KjellMorgenstern, 2019-10-21:
The version ca1a1bfd passed the tests on Fedora , travis job https://travis-ci.org/fritzing/fritzing-app/jobs/600260286 .

That means, it compiled, linked and the binary was successful run to process the parts database.

cmbahadir, 2019-10-21:
Hi @tinpotnick ,

Path of the libraries are different on almost all linux distros. I have also faced a linking problem on my Linux Mint. Here the issue i have openned and closed afterwards #3543 .

For a local build, IMO, it is safe to add it to Makefile but i would suggest to add them to .pro file as in #3544 .  Since, yes, as you said, Makefiles are autogenerated with qmake. So, first you should add it to phoenix.pro and then run qmake again to create the Makefiles.

I do not know about a generic development script to support all Linux distros which may not be possible for now.

KjellMorgenstern, 2019-10-21:
It would be nice if you could check this binary. It was build on and for Fedora 30. It was tested as stated above, but since I don't run any Fedora, i can not to extensive testing.

https://github.com/fritzing/fritzing-app/releases/download/CD-415/fritzing-8f47c89054fdb85a098e2c2253e3e37573618628-master-415.fedora-30.linux.AMD64.tar.bz2

tinpotnick, 2019-10-22:
Hi @KjellMorgenstern, I just ran the version you linked to and it all runs well (it is a version behind though? 0.9.4 rather than 0.9.5?).

KjellMorgenstern, 2019-10-22:
@tinpotnick It is from the master branch, build 415. So, yes that is 0.9.4. On the development branch it should show 0.9.5**d** . I just noted that the format is a bit messed up "Version 0.9.5
(dCD-380-..."  , with the build type moved away from the version number.

The hash from the devel branch,  ca1a1bfd , which you are using also builds fine on Fedora 30, like I mentioned above. That means it is likely that your Fedora 30 installations differs in some important point. Most likely, you are using some self build libraries instead of the Fedora 30 packages. See Dockerfile.fedora for the exact setup we are using to build.

closed 2019-10-23 by KjellMorgenstern