fritzing/fritzing-app #3548: Not able to download fritzing from homepage

KilianB, 2019-10-23:
I tried to download fritzing on a secondary device and can not find the link to download the packages. 

Also the pay nothing selection vanished.
Tested on Chrome and Firefox with and without adblocker

[image]

KjellMorgenstern, 2019-10-23:
For everybody who is just here for the download: **Please sign up / log in** on the webpage to download again :-)

For the greater picture.
The "free beer" download missing is not a bug. Fritzing almost died. The server was infested with bots that tried to hack other websites. The email system was not working anymore. Email reputation down, probably it was spamming, too, but I don't have proof for that. Google capture plugin ran out, and it took several months to fix this. Valid bug fixes and pull requests from the community were ignored. Knowledge about the build toolchain lost, dependency on unsupported tools, website not updated to comply with law, website missing Transport Layer Security, Spam and viruses on the Fritzing project page, obsolete tickets staying open forever... I could go on longer. What I am trying to say is that unfixed bugs and an outdated release 0.9.3b was only the tip of the iceberg. 

My working assumption is that this was because developers prefer paid work, and that Fritzing is too complex to be maintained as a hobby. It never was some developer's side project: There was public funding and other sources, which are not available anymore.

I think most users are happy to pay some money if it is used for the further development of the application. How exactly this will look like is not yet figured out. Paying for a "download service" is unlikely to be the long term solution, although currently this even works, due to the immense momentum this project still has :1st_place_medal:  .

Although I think the original topic is solved (not able to download), I am leaving this open for further comments.

WimberleyTech, 2019-10-23:
What is currently the best way to donate $$ to the project?  I will make a donation today :-)

KjellMorgenstern, 2019-10-23:
@WimberleyTech Private gifts can be send to kjell@gmx.de via Paypal. 
However I wonder if I should talk to a tax advisor if it is ok to ask for private gifts in public. Maybe one is reading this and would like to donate an advise :-) ? You can also directly go to https://fritzing.org/donate . This will currently also end up at my disposal, and would be a more official form.

wraybowling, 2019-10-29:
@KjellMorgenstern I'd like to see a Patreon. It's how I donate monthly to Libretro for example. (https://www.patreon.com/libretro)

KilianB, 2019-10-29:
Patreaon is a good idea. 

Just to throw in one more alternative; I am not entirely sure if this is just a github beta feature, but somewhere you may opt in to allow people to donate to repositories directly on github. During the first year every single dollar received will be matched as well. https://github.com/sponsors

wraybowling, 2019-10-29:
That's awesome! Heck, do both.

KjellMorgenstern, 2019-10-29:
I might try Patreon at some time, but after a first check they seem very expensive. 
5% Patreon + 1% extra for Paypal from Outside US + 2.9% + 30cent Paypal + 2.5% conversion to $ + 2.5% conversion to €  = 12.9% + 30 cent. Additional 20$/month for monthly payout. And Fritzing only needs payment processing, not a webpage or anything else of their services as far as I can tell (I might be wrong about that?)
In the mid/long term, I will add a US specific solution. About 20% of the users are from the US, which is the biggest user group. Note that fritzing.org/donate already offers $ payment.

@wraybowling It often is not recommended to open accounts at different collectors, as users might wonder which one is the correct official link, and wander away.

wraybowling, 2019-10-30:
Well within 6 hours Github approved me for their new sponsor feature. Don’t let the waitlist scare you off.

Rummaging through their docs I’m not finding any information on how much money they will take after the dollar matching. https://help.github.com/en/github/supporting-the-open-source-community-with-github-sponsors/becoming-a-sponsored-developer#joining-github-sponsors

KilianB, 2019-10-30:
> In the first year, GitHub will not charge any fees, so 100% of sponsorships will go to the sponsored developer. In the future, we may charge a nominal processing fee.
https://help.github.com/en/github/supporting-the-open-source-community-with-github-sponsors/about-github-sponsors

DavidBijttebier, 2019-11-05:
Very sorry to hear the project almost died. 
I have not used it a lot yet but I think I like it and might consider donating as well. 
(But I think big company's such as Sparkfun, Adafruit, Velleman, Arduino, Raspberry pi,.... maybe even Aliexpres or Banggood should fund this project since they would obviously also benefit from keeping it alive...)

But the solution to not paying right now does not work for me since I have not received a confirmation mail after signing up (neither in spam or other folder...). 
And when I try to login I get the message that my account is inactive.
Trying too sign up again does not work since my email has has allready been used...  
I only tried a gmail account. 

(Also the website is not https :) )

KjellMorgenstern, 2019-11-05:
@DavidBijttebier In case you can not log in, the webpage should show you the following message:
"If you have any trouble logging in to your account, please contact us.", with a link to out contact page: https://fritzing.org/contact/ .

The setup currently supports both, http and https. Setting a global permanent redirect is not yet possible.

marekkalnik, 2019-11-18:
Hi, it's great that you are working on finding a sustainable model to improve the project!
However, the website wording is a bit confusing:

> Fritzing is devoted to making creative use of electronics accessible to everyone.
> 
> Thousands of hours are required each year to keep up the development. Please consider to fund Fritzing before downloading the app.

I understood it a bit like "here's the free version, but you can also pay", but the free version is nowhere to be found. It would be nice to either direct the user to GH or make clear that paying is not optional if you want to download from the website.

deshipu, 2019-11-18:
There used to be a $0 option in there.

KilianB, 2019-11-19:
> For everybody who is just here for the download: **Please sign up / log in** on the webpage to download again :-)

The very first answer tells you how to download "for free". Just login in to the website and you will be able to download the program.

OMGdaDPS, 2019-11-29:
right on, thank you.

ccoenen, 2020-03-05:
Projection: this will scare off anyone getting started with electronics. They will probably just download fusion 360 (which now includes an eagle license for private use) or something like that. At some point, the community will just die. And then there will be nothing.

I still have fritzing on my machine, to clicking the update-button just brought me to http://fritzing.org/download/api/1.0/update/344 where I could get 0.9.4 for windows. But anyone just starting out does not have that - and will likely skip Fritzing altogether.

KjellMorgenstern, 2020-03-05:
@ccoenen Fritzing already was at "nothing" and many different income and side funding methods have been tried. But I am just repeating what I already wrote above. Wether users will turn away or not, or better, if this method will be successful for the next years, we will see. I don't see how your projection helps with this topic.

smalltimebloke, 2020-04-03:
Kjell, Have you considered adding a different payment method? Liberapay is a good bet.