fritzing/fritzing-app #3550: No Windows 10 builds available from latest release

blobkat, 2019-10-23:
Hello, I see that there's CI set up for Windows, however I don't see any Fritzing 0.9.4 binaries for Windows to download anywhere?

closed 2019-10-23 by KjellMorgenstern

KjellMorgenstern, 2019-10-23:
To give a bit more detail:
* The Windows build needed manual afterworks (Windows choked on the long file names, partially unzipped, and then Fritzing simply crashed after the user ignored the warning about not extracting all files... way to get in lots of support requests)
* To minimize the potential risk, I release to a smaller group first. Sorry Linux people for making you guinea pigs. 
* Some further obstacles just vanished, maybe I can finish it today. But I thought that now three days in a row :-)

blobkat, 2019-10-23:
OK! Superb, no worries, I'm just excited that the Fritzing project is alive again :) Thank you for all the good work you're doing.