fritzing/fritzing-app #3551: Windows and MacOS binarys missing in release CD415

vanepp, 2019-10-23:
As reported in forum post 

https://forum.fritzing.org/t/windows-and-mac-developer-downloads-9-1-4-not-available/7540

The MacOs and Windows binaries aren't in CD415 the 0.9.4 release. I assume this is a mistake rather than intentional?

Peter

WoutStandaert, 2019-10-24:
Yesterday I asked the same question with my other account: https://github.com/fritzing/fritzing-app/issues/3550

vanepp, 2019-10-25:
Thanks! I didn't see that nor the explainition, because 3550 has been closed already ... Any one else interested, just search for issue 3550, or be assured it is being worked on which is all I care about. 

Peter

shfitz, 2019-11-07:
I tried compiling the OSX version myself the other day, but with no luck- any Mac binaries for the latest release available?

vanepp, 2019-11-07:
While I don't know the details (I don't have a Mac), I believe they had to  static link libgit2 on the Mac to get it to work (Linux works out of the box and I had a year and a half of trouble getting libgit2 to compile and link on Windows.) The binaries for Windows and the Mac are on the todo list but obviously haven't hit the top yet. There are many things wrong including the web site, and only a very limited amount of staff time (and I don't have access or the skills to help with any of that.) So far there are only a few new fixes in 0.9.4 release, it is mostly a test of the tool and distribution chain, so 0.9.3B is available (although I think there is one fix for one Mac version that is only in 0.9.4, if that is what you need) so that may or may not do for you. Hopefully @KjellMorgenstern  will provide a better update!

Peter

KjellMorgenstern, 2019-11-07:
Thanks for the explanation Peter, that describes the current situation. We are making progress behind the scenes, binaries are really high on the list.

alastaira, 2019-12-02:
Windows binaries still seem to be missing from the CD-498 release. Is there anything useful a "layperson" can do to help support this fantastic effort? (perhaps just offer words of encouragement about how pleased I was to see Fritzing development re-started? Go team!)

KjellMorgenstern, 2019-12-02:
@alastaira Yes, the binaries are out since yesterday, including Windows 64 Bit. There is something very simple for the layperson: Use the paid download at https://fritzing.org 

The alternative, getting involved is with Fritzing development, is quite a task, and I think it is a lot to ask for if you just want to use the application. Words are very welcome, but my strategy is to emphasize that Fritzing is Open Source, while coffee, computers, webservices, developers, rent, ..., are usually not. Fritzing had different approaches for finance that worked well in the past, but not for the last few years.

WimberleyTech, 2019-12-02:
I have made several donations so far (and will likely continue to
    make additional ones) and I STRONGLY encourage others to do so as
    well.
    
    doug
    
    On 12/2/2019 5:41 AM, Kjell wrote:
    
    
      
      @alastaira
        Yes, the binaries are out since yesterday, including Windows 64
        Bit. There is something very simple for the layperson: Use the
        paid download at https://fritzing.org
      The alternative, getting involved is with Fritzing development,
        is quite a task, and I think it is a lot to ask for if you just
        want to use the application. Words are very welcome, but my
        strategy is to emphasize that Fritzing is Open Source, while
        coffee, computers, webservices, developers, rent, ..., are
        usually not. Fritzing had different approaches for finance that
        worked well in the past, but not for the last few years.
      —
        thread.
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/fritzing/fritzing-app/issues/3551?email_source=notifications\u0026email_token=ALKDFNPGCHKHVCJNUNTIYCTQWTX5LA5CNFSM4JEJDNH2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFTGXFQ#issuecomment-560360342",
"url": "https://github.com/fritzing/fritzing-app/issues/3551?email_source=notifications\u0026email_token=ALKDFNPGCHKHVCJNUNTIYCTQWTX5LA5CNFSM4JEJDNH2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFTGXFQ#issuecomment-560360342",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

EtianAM, 2019-12-14:
If you have Fritzing 0.9.3, you can update a download the binaries of windows,

failiz, 2021-03-09:
Outdated

closed 2021-03-09 by KjellMorgenstern