fritzing/fritzing-app #3558: Crashing if trying to route the PCB

NilsRo, 2019-11-02:
Hi,

I saw issues with wires on my growing project, that deleting it in the Breadboard view does keep it in the others and so on. After restarting Fritzing it is correct than. Also I cannot route my project in PCB view. Fritzing instantly crashes. The project is attached.

Regards,
Nils
[Sicherheitsschaltung Ladekoffer.zip](https://github.com/fritzing/fritzing-app/files/3800803/Sicherheitsschaltung.Ladekoffer.zip)

KjellMorgenstern, 2019-11-02:
Did github show you the bug reporting template when you created this issue?
I am asking because without the Fritzing version, this report is near useless.

NilsRo, 2019-11-02:
Hi,

did not see it, perhaps I had to log in before creating it. But now I can see the template if I try to create a new issue.

## Current Behaviour
**Build:**
Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.6.1]

**Operating System:**
Windows 10 64Bit

**Steps to reproduce:**
- Try Autoroute in PCB view

## Expected Behaviour
No crash

Cheers,
Nils

KjellMorgenstern, 2019-11-02:
Ah, thanks. Did you use a link from Fritzing to report the issue (that would mean I'd have to fix the link)?

A lot of those crashes have been fixed in 0.9.4, which I hopefully will be able to publish soon (its already finished and published for Linux)

NilsRo, 2019-11-02:
Yes, I used it. But now it is working fine. Seems to be related that I was not logged in before.

Did you tested it there or should I wait for the new version?

Cheers,
Nils

KjellMorgenstern, 2019-11-02:
Fritzing works offline, no relation to login. It’s best if you can retry this when 0.9.4 is available. I am really sorry i can not publish yet.

NilsRo, 2019-11-03:
Yes, but Guthub did not. I had to look for my login. Did you have an estimation for the releasedate? If it is really far away you perhaps can try it with linux version if it is still a bug? If not we can close the issue here.

KjellMorgenstern, 2019-11-03:
No crash on my current development build.
Of course the autorouter will only parts which are aligned on the PCB, which some of your parts are not yet, so many connections are missing.
[image]

closed 2019-11-03 by NilsRo