fritzing/fritzing-app #3559: Drawing a connection with ALT for connection a part moves the source wire

NilsRo, 2019-11-02:
## Current Behaviour
Drawing a connection with ALT for connection a part moves the source wire. See video for a detailed description. 
[2019-11-02_20-42-51.zip](https://github.com/fritzing/fritzing-app/files/3800894/2019-11-02_20-42-51.zip)

**Build:**
Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.6.1]

**Operating System:**
Windows 10 64 Bit

**Steps to reproduce:**
Drawing a connection with ALT between a wire and a part

## Expected Behaviour
The source wire should now change it arragement.

NilsRo, 2019-11-03:
Yes, I used ALT for creating a connection from the wire. Today I also recognized that there is the same behavior when creating new bending point.

failiz, 2021-11-05:
Related to #3738 #3602 and #1869. I made a fix for this issue on my PR.

commit ade78caa (2022-06-01): fixes the moving of the bendpoint when draging a wire from the bendoint (using the ALT key). Fixes #3559
closed 2022-06-01 by KjellMorgenstern