fritzing/fritzing-app #3569: Selecting overlapping parts

KjellMorgenstern, 2019-11-25:
## Problem

it is difficult and almost impossible to "click" the part you want if they are overlapping (best thing: IC on top and a SMD resistor on the bottom, try to catch the resistor! - no way)
rectangular PCB

## Proposed Solution
?

deshipu, 2019-11-25:
I think it would become much less of a problem if:

1. "bring to front" and "send to back" actually worked, and
2. selecting parts respected their order in those layers, *and* was consistent with highlighting (which it is not at the moment, often you will have one part highlighted on mouse over, and a different part selected when you click).

Then the solution to "I get the wrong part selected" is to simply select it and send it to back, repeating that operation until we select the part that we need. However, this wouldn't always work if we wanted to select multiple parts at once (holding down ctrl), with some of them possibly overlapping.

failiz, 2021-03-25:
#52 could mitigate this in some cases
Another way of fixing this would create a contetual menu that appears when there are several selection shapes overlapping and lets you decide which one you want (this is what Kicad and other CAD programs do).
It could also be possible to select the part that has the minimum distance to one of its connectors. So, if you click to nearby of one part connectors, that part will be selected.

deshipu, 2021-03-25:
I find it that can often select the part I want by right-clicking on it and then dismissing the context menu, however, this selection is somehow different than selecting it with a left click — for instance, arrow keys will then scroll the view instead of moving the part. It would help to make those selections both behave the same.

KjellMorgenstern, 2021-03-26:
Related / duplicate of #943

failiz, 2021-03-26:
Add bug and pcbview labels

KjellMorgenstern, 2021-06-15:
Summary of suggested solutions
a. Mouse wheel selection
b. Consider exact part shapes to reduce overlap
c. Fix mouse over highlighting, sometimes it is not in line with the part that gets selected
d. Fix bring to front / sent to back, it is not reliable

Actually, all of those could and should be used.