fritzing/fritzing-app #3575: Having trouble running Fritzing on Linux Mint 19.2.

ehoitinga, 2019-12-10:
## Current Behaviour
Having trouble running Fritzing on Linux Mint 19.2.

See: https://forum.fritzing.org/t/trouble-installing-fritzing-on-linux-mint-19-2/7881

Screenshots: http://users.skynet.be/fa403916/Screenshots-Errors.pdf
<please attach screenshot if possible>
**Build:**
0.9.4

**Operating System:**
Linux laptop-erik 5.3.0-24-generic #26~18.04.2-Ubuntu SMP Tue Nov 26 12:34:22 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

**Steps to reproduce:**
Unzip Fritzing tar ball
Run ~/bin/fritzing-0.94/Fritzing
Click OK on error messages until the app appears
Click "Breadboard"
Click "Part --> Regenerate part database"
Click "Regenerate"
Click "Ok" on the error
Click "File --> Quit"

**Terminal output:**
```
erik@laptop-erik:~/bin/fritzing-0.94$ ./Fritzing
qt5ct: using qt5ct plugin
QSqlDatabase: QSQLITE driver not loaded
QSqlDatabase: available drivers: 
QSqlDatabase: QSQLITE driver not loaded
QSqlDatabase: available drivers: 
QSqlDatabasePrivate::removeDatabase: connection 'qt_sql_default_connection' is still in use, all queries will cease to work.
QSqlDatabasePrivate::addDatabase: duplicate connection name 'qt_sql_default_connection', old connection removed.
QSqlDatabase: QSQLITE driver not loaded
QSqlDatabase: available drivers: 
QSqlDatabasePrivate::removeDatabase: connection 'qt_sql_default_connection' is still in use, all queries will cease to work.
QSqlDatabasePrivate::addDatabase: duplicate connection name 'qt_sql_default_connection', old connection removed.
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
QSqlQuery::prepare: database not open
qt5ct: D-Bus global menu: no
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread

 [Snipped numerous identical lines]

QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QSqlDatabase: QSQLITE driver not loaded
QSqlDatabase: available drivers: 
QSqlDatabasePrivate::removeDatabase: connection 'qt_sql_default_connection' is still in use, all queries will cease to work.
QSqlDatabasePrivate::addDatabase: duplicate connection name 'qt_sql_default_connection', old connection removed.
QSqlDatabase: QSQLITE driver not loaded
QSqlDatabase: available drivers: 
QSqlDatabasePrivate::removeDatabase: connection 'qt_sql_default_connection' is still in use, all queries will cease to work.
QSqlDatabasePrivate::addDatabase: duplicate connection name 'qt_sql_default_connection', old connection removed.
QSqlDatabase: QSQLITE driver not loaded
QSqlDatabase: available drivers: 
QSqlDatabasePrivate::removeDatabase: connection 'qt_sql_default_connection' is still in use, all queries will cease to work.
QSqlDatabasePrivate::addDatabase: duplicate connection name 'qt_sql_default_connection', old connection removed.
erik@laptop-erik:~/bin/fritzing-0.94$
```

## Expected Behaviour
App should start normally without errors :-)

KjellMorgenstern, 2019-12-10:
Can you try to install additional qt libraries:
`sudo apt install libqt5xml5 libqt5sql5 libqt5serialport5`

Also, can you check the content of $HOME/.config/Fritzing/Fritzing.conf .
There is a [bins2] section, what is the path for location=user and location=app ?

ehoitinga, 2019-12-10:
Apparantly these packages were already installed.

```
erik@laptop-erik:~/bin$ sudo apt install libqt5xml5 libqt5sql5 libqt5serialport5
[sudo] password for erik:            
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libqt5serialport5 is already the newest version (5.9.5-0ubuntu1).
libqt5sql5 is already the newest version (5.9.5+dfsg-0ubuntu2.4).
libqt5xml5 is already the newest version (5.9.5+dfsg-0ubuntu2.4).
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
erik@laptop-erik:~/bin$
```
The "7 not upgraded" are samba packages which have nothing to do with Fritzing. Below the contents of ~/.config/Fritzing/Fritzing.conf:

```
0\location=user
0\path=/home/erik/Documents/Fritzing/bins/search.fzb
0\title=Search

1\location=app
1\path=/home/erik/bin/fritzing-0.94/fritzing-parts/bins/core.fzb
1\title=Core Parts

12\location=app
12\path=/home/erik/bin/fritzing-0.94/fritzing-parts/bins/contribParts.fzb
12\title=Contributed Parts
```

```
erik@laptop-erik:~/bin$ ls -l /home/erik/Documents/Fritzing/bins/search.fzb
-r--r----- 1 erik erik 160 déc 10 18:08 /home/erik/Documents/Fritzing/bins/search.fzb
erik@laptop-erik:~/bin$ ls -l /home/erik/bin/fritzing-0.94/fritzing-parts/bins/core.fzb
-rw-r--r-- 1 erik erik 48409 déc  1 14:24 /home/erik/bin/fritzing-0.94/fritzing-parts/bins/core.fzb
erik@laptop-erik:~/bin$ ls -l /home/erik/bin/fritzing-0.94/fritzing-parts/bins/contribParts.fzb
-rw-r--r-- 1 erik erik 2976 déc  1 14:24 /home/erik/bin/fritzing-0.94/fritzing-parts/bins/contribParts.fzb
erik@laptop-erik:~/bin$

```
Erik

KjellMorgenstern, 2019-12-10:
This looks all good so far.
The install of those three libraries should also have triggered installing the
libqt5sql5-sqlite library. Probably it is also there, but better check,since the error message is related to that.

Another thing to try
`ls -l /home/erik/bin/fritzing-0.94/fritzing-parts/parts.db`

The file should have +rw permission for the user, and be about 11MB. If the file has zero size Fritzing will detect this and recreate it from the parts. But if the file is broken, lets say only 1MB, Fritzing might not be able to recover. It is save to remove the file. Fritzing will automatically regenerate it on the next start.

ehoitinga, 2019-12-10:
The libqt5sql5-sqlite library was not there. Installing it with `sudo apt install libqt5sql5-sqlite` did the trick. Fritzing starts up without any GUI errors. 

So for now on the following packages should be present on linux mint 19.2 before installinh Fritzing:

```
sudo apt-get update
sudo apt-get upgrade
sudo apt install libqt5printsupport5 libqt5xml5 libqt5sql5 libqt5serialport5 libqt5sql5-sqlite
```

Below the terminal output. Don't know if I should be worried about that. Will do furter testing tomorrow.

```
erik@laptop-erik:~/bin/fritzing-0.94$ ./Fritzing
qt5ct: using qt5ct plugin
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
qt.svg: link B hasn't been detected!
qt.svg: link C hasn't been detected!
qt.svg: link D hasn't been detected!
qt.svg: link E hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link C hasn't been detected!
qt.svg: link D hasn't been detected!
qt.svg: link E hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link C hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link C hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link D hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link E hasn't been detected!
qt.svg: link F hasn't been detected!
qt.svg: link E hasn't been detected!
qt.svg: link D hasn't been detected!
qt.svg: link E hasn't been detected!
qt.svg: link F hasn't been detected!
qt.svg: link G hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link G hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link C hasn't been detected!
qt.svg: link B hasn't been detected!
qt.svg: link C hasn't been detected!
qt.svg: link D hasn't been detected!
qt.svg: link D hasn't been detected!
qt.svg: link D hasn't been detected!
qt.svg: link E hasn't been detected!
qt.svg: link F hasn't been detected!
qt.svg: link G hasn't been detected!
qt.svg: link H hasn't been detected!
qt.svg: link G hasn't been detected!
qt.svg: link I hasn't been detected!
qt.svg: link J hasn't been detected!
qt.svg: link J hasn't been detected!
qt.svg: link I hasn't been detected!
qt.svg: link E hasn't been detected!
qt.svg: link F hasn't been detected!
qt.svg: link K hasn't been detected!
qt.svg: link H hasn't been detected!
qt.svg: link I hasn't been detected!
qt.svg: link K hasn't been detected!
qt.svg: link K hasn't been detected!
qt.svg: link I hasn't been detected!
qt.svg: link L hasn't been detected!
qt.svg: link L hasn't been detected!
qt.svg: link M hasn't been detected!
qt.svg: link N hasn't been detected!
qt.svg: link O hasn't been detected!
qt.svg: link N hasn't been detected!
qt.svg: link P hasn't been detected!
qt.svg: link Q hasn't been detected!
qt.svg: link P hasn't been detected!
qt.svg: link M hasn't been detected!
qt.svg: link O hasn't been detected!
qt.svg: link Q hasn't been detected!
qt.svg: link R hasn't been detected!
qt.svg: link R hasn't been detected!
qt.svg: link S hasn't been detected!
qt.svg: link T hasn't been detected!
qt.svg: link T hasn't been detected!
qt.svg: link U hasn't been detected!
qt.svg: link V hasn't been detected!
qt.svg: link W hasn't been detected!
qt.svg: link X hasn't been detected!
qt.svg: link S hasn't been detected!
qt.svg: link T hasn't been detected!
qt.svg: link T hasn't been detected!
qt.svg: link U hasn't been detected!
qt.svg: link V hasn't been detected!
qt.svg: link W hasn't been detected!
qt.svg: link X hasn't been detected!
qt.svg: link Y hasn't been detected!
qt.svg: link Z hasn't been detected!
qt.svg: link Y hasn't been detected!
qt.svg: link Z hasn't been detected!
qt.svg: link Y hasn't been detected!
qt.svg: link Z hasn't been detected!
qt.svg: link Z hasn't been detected!
qt5ct: D-Bus global menu: no
erik@laptop-erik:~/bin/fritzing-0.94$
```

Erik

vanepp, 2019-12-10:
The above looks pretty normal for console output on Linux, it seems to work fine anyway. It would be nice to eliminate the error messages, but bugs that break things are a much higher priority for limited developer time. @KjellMorgenstern  should we need to add those libraries? I thought on 0.9.3b all the Qt libraries needed were in the tarball (although I may be wrong there.) Are the Qt libraries a standard part of the Linux distros these days?

Peter

ehoitinga, 2019-12-11:
On my fresh linux installation in Virtualbox. The QT5 libraries are installed. However QT itself is, I think, not installed:

```
erik@mint19-vm01:~$ qmake -v

Command 'qmake' not found, but can be installed with:

sudo apt install qtchooser

erik@mint19-vm01:~$ sudo apt install qtchooser
[sudo] password for erik:            
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  qtchooser
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 24,1 kB of archives.
After this operation, 125 kB of additional disk space will be used.
Get:1 http://ftp.belnet.be/ubuntu bionic/main amd64 qtchooser amd64 64-ga1b6736-5 [24,1 kB]
Fetched 24,1 kB in 0s (212 kB/s)     
Selecting previously unselected package qtchooser.
(Reading database ... 290708 files and directories currently installed.)
Preparing to unpack .../qtchooser_64-ga1b6736-5_amd64.deb ...
Unpacking qtchooser (64-ga1b6736-5) ...
Setting up qtchooser (64-ga1b6736-5) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
erik@mint19-vm01:~$ qmake -v
qmake: could not find a Qt installation of ''
erik@mint19-vm01:~$ 
```
As far as I can see the QT5 libraries are in `/usr/lib/x86_64-linux-gnu/qt5` and `/usr/share/qt5/`.

In stead of including the necessary libraries in the tarball maybe it is worthwhile to create a PPA and handle dependencies there?

Will continue to test and if you want I can continue to look in to the installation in Linux Mint by taking a look at the install_fritzing.sh script to find out which lines are necessary to install the mime types and to get Fritzing in the menu.

Let me know.

Erik

KjellMorgenstern, 2019-12-11:
If the user did not install any other app that depends on qt5, they will run into this error.
It seems that some users ran into trouble with the install script. For now I created instructions INSTALL.txt and removed the install script, as it seems to do more harm than help.

https://github.com/fritzing/fritzing-app/commit/091ff39ce380e2e6e4f5d5e2acb8b934a0bb1541

ehoitinga, 2019-12-11:
@KjellMorgenstern On my fresh VM install of linux mint 19.2 I also had to install libqt5printsupport5 and libqt5sql5-sqlite.

To be on the safe side maybe it is better to use the command `sudo apt install libqt5printsupport5 libqt5xml5 libqt5sql5 libqt5serialport5 libqt5sql5-sqlite` in stead of `sudo apt install libqt5xml5 libqt5sql5 libqt5serialport5` before starting Fritzing.

Erik

KjellMorgenstern, 2019-12-11:
I'll add this and also instructions for fedora. By the way, removing the scripts doesn't mean I want to dump them, but currently I just don't have time to verify and improve them.

vanepp, 2019-12-11:
I think that the release script for 0.9.3b copies all the Qt libraries in to the tar ball as this from 

https://github.com/fritzing/fritzing-app/wiki/4.-Publishing-a-Release

says

"...
Linux 64 bit

Official Qt documentation: http://doc.qt.io/qt-5/linux-deployment.html

There is a good set of instructions for setting up virtual machines and building the linux releases here, so I will just add a bit of context. First, we do our builds under Ubuntu 16.04 LTS. Basically, to do the build, you use the script release.sh, and it can be found in Fritzing source in tools/linux_release_script/. Once your vm is set up correctly, you merely have to run the script with the current release string as the parameter (e.g.: ./release.sh 0.8.6b). The process will create a standalone tar.bz2 file containing the Fritzing executable plus all the necessary auxiliary files (including the qt libraries). These bz2 files are download-ready."

I know this because if you don't run this script (or make a sym link for libgit2, which is what I used to do) when building from source for 0.9.3b, it won't link libgit2 properly. Running the release script fixed that. Here is an ls from the 0.9.3b tarball on Linux (Ubuntu 18.04 lts):

fritzing-0.9.3b.linux.AMD64/lib:
Fritzing
imageformats
libgit2.so
libgit2.so.0.23.4
libgit2.so.23
libicudata.so.56
libicudata.so.56.1
libicui18n.so.56
libicui18n.so.56.1
libicuuc.so.56
libicuuc.so.56.1
libQt5Concurrent.so
libQt5Concurrent.so.5
libQt5Concurrent.so.5.6
libQt5Concurrent.so.5.6.0
libQt5Core.so
libQt5Core.so.5
libQt5Core.so.5.6
libQt5Core.so.5.6.0
libQt5DBus.so
libQt5DBus.so.5
libQt5DBus.so.5.6
libQt5DBus.so.5.6.0
libQt5Gui.so
libQt5Gui.so.5
libQt5Gui.so.5.6
libQt5Gui.so.5.6.0
libQt5Network.so
libQt5Network.so.5
libQt5Network.so.5.6
libQt5Network.so.5.6.0
libQt5PrintSupport.so
libQt5PrintSupport.so.5
libQt5PrintSupport.so.5.6
libQt5PrintSupport.so.5.6.0
libQt5SerialPort.so
libQt5SerialPort.so.5
libQt5SerialPort.so.5.6
libQt5SerialPort.so.5.6.0
libQt5Sql.so
libQt5Sql.so.5
libQt5Sql.so.5.6
libQt5Sql.so.5.6.0
libQt5Svg.so
libQt5Svg.so.5
libQt5Svg.so.5.6
libQt5Svg.so.5.6.0
libQt5Widgets.so
libQt5Widgets.so.5
libQt5Widgets.so.5.6
libQt5Widgets.so.5.6.0
libQt5XcbQpa.so
libQt5XcbQpa.so.5
libQt5XcbQpa.so.5.6
libQt5XcbQpa.so.5.6.0
libQt5XmlPatterns.so
libQt5XmlPatterns.so.5
libQt5XmlPatterns.so.5.6
libQt5XmlPatterns.so.5.6.0
libQt5Xml.so
libQt5Xml.so.5
libQt5Xml.so.5.6
libQt5Xml.so.5.6.0
platforms
sqldrivers

fritzing-0.9.3b.linux.AMD64/lib/imageformats:
libqjpeg.so

fritzing-0.9.3b.linux.AMD64/lib/platforms:
libqxcb.so

fritzing-0.9.3b.linux.AMD64/lib/sqldrivers:
libqsqlite.so

On .9.3b the only times I've seen install problems on Linux over the years is when the user used one of the package managers to install Fritzing (some of them omitted the parts repository which caused problems), in all cases I know of uninstalling that version and installing the distribution tarball instead just worked (I think because all the required parts are in lib although I could be wrong.) I expect we would have had more complaints if it was depending on Qt being installed already because of systems that didn't have Qt installed.

KjellMorgenstern, 2020-02-04:
Updated instructions on qt5 dependencies in INSTALL.txt

closed 2020-02-04 by KjellMorgenstern

h1nver, 2020-02-17:
sudo apt-get update
sudo apt-get install libqt5printsupport5
apt-get install libqt5serialport5-dev

Try this, sorry I can't speak English