fritzing/fritzing-app #3579: No routing complete message when no wires used to connect parts

vanepp, 2019-12-15:
## Current Behaviour

With somewhat non standard parts (with pins configured as female to connect directly to a part with male connectors without wires), the routing complete message breaks. Routing is complete, but is not reported so:

[image]

The bottom two connections are in fact correct (they appear as rats nest lines in schematic and pcb) but are unusual in that they are a male to female connection without a wire (the top two are connected with wires to display correct behaviour.  More information on these parts is available here:

https://forum.fritzing.org/t/part-creation-howto-part5-3d-images-howto/7916

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

Version 0.9.4 (CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
<Windows 7>

**Steps to reproduce:*

- load this sketch in to Fritzing (after removing the trialing .zip):

[not-routed.fzz.zip](https://github.com/fritzing/fritzing-app/files/3965280/not-routed.fzz.zip)

## Expected Behaviour

The routing complete message should appear in place of the 4 nets to be routed message.

failiz, 2020-11-17:
I think that this issue has been fixed with #3727. In current 0.9.5, the breadboard view reports "routing completed"

vanepp, 2021-03-13:
Indeed fixed in 0.9.6 so I will close this! Thank you!

closed 2021-03-13 by vanepp