fritzing/fritzing-app #3580: Wires in schematic view are not drawn correctly if part is moved

NilsRo, 2019-12-16:
## Current Behaviour
In schematic view the wires are not drawn correctly if a parted is moved. It seems that the are not connected anymore but it is still fine. Only the wires drawn on the monitor are not correct anymore.

[image]

Video: https://readycloud.netgear.com/client/dllink.html#t=01h4idsbrhl2raau8xq76wxzkij/2019-12-16_22-30-06.mp4

[Sicherheitsschaltung Ladekoffer.zip](https://github.com/fritzing/fritzing-app/files/3970246/Sicherheitsschaltung.Ladekoffer.zip)

**Build:**
Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Move parts in schematics in the attached circuit.

## Expected Behaviour
Wires drawn correctly.

KjellMorgenstern, 2019-12-16:
A related problem exists in breadboard view.
Note: the displayed number of unrouted connections is probably not related to this issue (see #3581) .

NilsRo, 2019-12-17:
Hi, #3581 is not related to this. Today I ran into the problem that the drawing cannot be fixed and I cannot connect a wire anymore. 
[image]
[Sicherheitsschaltung Ladekoffer.zip](https://github.com/fritzing/fritzing-app/files/3973139/Sicherheitsschaltung.Ladekoffer.zip)

KjellMorgenstern, 2019-12-17:
When I click the connector 1 of R32, it is highlighted in yellow, together with R31.1 , C12.2 and all the others of this subnet.  Further, if I move the bendpoint left of R32, then R32 might jump as well.

KjellMorgenstern, 2019-12-17:
Created a smaller version of the file that still has/triggers this error:
[simplified-3580.fz.zip](https://github.com/fritzing/fritzing-app/files/3974731/simplified-3580.fz.zip)

NilsRo, 2020-07-26:
I do not think #3608 is related to this issue as it is a drawing issue on schematic view only. The wire was there and not drawn correctly after moving the parts anymore. The connection in the background is fine.

failiz, 2022-03-24:
> Is this not the same as #3006?
No, #3006 happens when dragging bendpoints and it can bi fixed with #3942 .
This issue is related to #3006 , but happens when dragging a part. There is a video of this behaviour in #3006. 
In any case, the problem is probably the same: the code does not find all the connected wires to a connector. The solution would be very similar to  #3942 but in a different part of the code.