fritzing/fritzing-app #3581: One connectore is left for routing but there is any visible in schematics

NilsRo, 2019-12-16:
## Current Behaviour
In schematics view there is one connectore left for routing but I cannot find any.

[image]

[Sicherheitsschaltung Ladekoffer.zip](https://github.com/fritzing/fritzing-app/files/3970284/Sicherheitsschaltung.Ladekoffer.zip)

**Build:**
Version 0.9.4
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Open fritzing files attached and view schematics

## Expected Behaviour
There should be not wire left for routing or it must be somehow become visible what is left.

KjellMorgenstern, 2019-12-16:
It is late at night so I might be wrong, but it looks obvious: part 4N35, connector 6 is not routed?

KjellMorgenstern, 2019-12-16:
In Menu -> Routing -> Show unrouted

It will tell "Unrouted connections are highlighted in yellow (I admit white on yellow is a bad choice).

[image]

KjellMorgenstern, 2019-12-16:
Note: The problem here is that R1 is only drawn upon the wire from R2 to C1, but no connection is set.

NilsRo, 2019-12-16:
Hi,

I also recognized now that if the dialog box in windows if confirmed
nothing is highlighted anymore. But with the dialog box no scrolling is
possible. :-)
Perhaps it would be nice to jump the the components not connected which
will be easier to use with larger schematics than looking for it manually.
Like with Next/Prev buttons.

Cheers and thanks for the late reply.

Nils

WimberleyTech, 2019-12-16:
I have duplicated the behavior on my system (also Win10).
    The net (node) stays intact even though it does not appear so
    graphically.
    If you delete the net in the schematic and then  double-click the
    resulting ratsnests, the bad behavior does not occur again.
    
    On the pcb view, the routing seems OK to me.
    
    Hope this helps.
    
    On 12/16/2019 3:47 PM, NilsRo wrote:
    
    
      
      Current Behaviour
      In schematics view there is one connectore left for routing but
        I cannot find any.
      
      Sicherheitsschaltung Ladekoffer.zip
      Build:
        Version 0.9.4
        (CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]
      Operating System:
        Windows 10
      Steps to reproduce:
      
        Open fritzing files attached and view schematics
      
      Expected Behaviour
      There should be not wire left for routing or it must be somehow
        become visible what is left.
      —
        thread.
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/fritzing/fritzing-app/issues/3581?email_source=notifications\u0026email_token=ALKDFNLB7VCGAM3HB7PVIGLQY7ZNXA5CNFSM4J3RXV4KYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IA3UBAA",
"url": "https://github.com/fritzing/fritzing-app/issues/3581?email_source=notifications\u0026email_token=ALKDFNLB7VCGAM3HB7PVIGLQY7ZNXA5CNFSM4J3RXV4KYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IA3UBAA",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

KjellMorgenstern, 2019-12-16:
@WimberleyTech I think you refer to issue #3580

WimberleyTech, 2019-12-17:
Correct. My bad.
    
    On 12/16/2019 5:11 PM, Kjell wrote:
    
    
      
      @WimberleyTech
        I think you refer to issue #3580
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/fritzing/fritzing-app/issues/3581?email_source=notifications\u0026email_token=ALKDFNNKVLNKYNWNTDG6KRLQZADLRA5CNFSM4J3RXV4KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHAOQJQ#issuecomment-566290470",
"url": "https://github.com/fritzing/fritzing-app/issues/3581?email_source=notifications\u0026email_token=ALKDFNNKVLNKYNWNTDG6KRLQZADLRA5CNFSM4J3RXV4KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHAOQJQ#issuecomment-566290470",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

NilsRo, 2019-12-18:
Not a real bug but could be optimized perhaps. But will close it.

closed 2019-12-18 by NilsRo