fritzing/fritzing-app #3582: Simple: Support CTRL-S for searching a component

NilsRo, 2019-12-18:
## Problem
Many times I am searching for a component I use CTRL-F but the shortcut is not supported in windows.

## Proposed Solution
Add shortcut for searching a component

KjellMorgenstern, 2019-12-19:
CTRL-S is a typo?

NilsRo, 2019-12-19:
Yep. CTRL-F. ;-)

janclod, 2021-08-20:
@NilsRo do you mean a shortcut to "Find part in sketch" from the menu "Part"? Or a shortcut to find a part from the "Search" tab inside the "Parts" dock (usually right side of the window)?

NilsRo, 2021-08-20:
> @NilsRo do you mean a shortcut to "Find part in sketch" from the menu "Part"? Or a shortcut to find a part from the "Search" tab inside the "Parts" dock (usually right side of the window)?

"Find part in sketch". I have a complex circuit and it would made it faster to work with it searching for the component on the circuit.

KjellMorgenstern, 2021-10-30:
Indeed, I missed setting the milestone.

closed 2021-10-30 by KjellMorgenstern