fritzing/fritzing-app #3584: GUI error: VIA selection not fine with overlapping parts

NilsRo, 2019-12-21:
## Current Behaviour
If a VIA is below a corner of another part it cannot be selected anymore sometimes. It is the case at the corners only.

[image]

**Build:**
CD-498-0-a-1ffcea 2019-12-01 64 Qt 5.12.3

**Operating System:**
Windows 10

**Steps to reproduce:**
- Put a VIA under the corner of another part.
- Try to select it.

## Expected Behaviour
Selection should be working in the same way as the VIA is not placed under the corner of another object.

[Untitled Sketch 3.zip](https://github.com/fritzing/fritzing-app/files/3991397/Untitled.Sketch.3.zip)

KjellMorgenstern, 2019-12-21:
There are various kind of issues like that.

NilsRo, 2020-11-25:
Any news about it? With compact PCBs it is nearly not possible to use fritzing as you have to move the overlapping component everytime. 
[image]

failiz, 2021-01-17:
related to #943

failiz, 2021-03-25:
Special case of  #3569. I think we should we move the info to that issue and close this one.

KjellMorgenstern, 2021-03-26:
I can not reproduce this exactly like in the screenshots, since the via has a larger selection rectangle around. However, if there are multiple conflicting areas around, it is still an issue.

Frtizing 0.9.6 , Left via is impossible to get, right via is easy:
[image]