fritzing/fritzing-app #3594: Variable Breadboard Wire Width

TSprech, 2020-01-13:
## Problem
I am developing a sizable library of large parts for use with Fritzing. An issue that presented itself along the way is the inability to change visual wire sizes in the breadboard view. The ability to change wire "gauge" or "mils" as it is defined in the save file, would be extremely helpful to convey more accurate visuals. I did not want to just request a feature without contributing some work towards the solution, so seen below is the testing I did and what I believe could remedy the problem.

## Proposed Solution
In an uncompressed .fz file composed of a simple breadboard and wires, a snippet of the code pertaining to "Wire1" can be seen:

`<title>Wire1</title>
            <views>
                <breadboardView layer="breadboardWire">
                    <geometry z="3.50002" x="279.001" y="153.001" x1="0" y1="0" x2="0" y2="-99.0006" wireFlags="64"/>
                    <wireExtras mils="22.2222" color="#25cc35" opacity="1" banded="0"/>
                    <connectors>`

For this enhancement, we are focusing on the `<wireExtras mils="22.2222"`

This results in the following standard Fritzing breadboard view:
[image]

My proposed solution is exposing the "mils" variable for users to edit in the Inspector menu, similar to how you would change resistance on a resistor or any other number field so the save file would look like this:
_(I changed the wire mils to 1/2 normal size, or from 22.2222 to 11.1111, however this works both for increasing and decreasing wire size as seen in the picture)_

`<title>Wire1</title>
            <views>
                <breadboardView layer="breadboardWire">
                    <geometry z="3.50002" x="243.001" y="153.001" x1="0" y1="0" x2="0" y2="-99.0006" wireFlags="64"/>
                    <wireExtras mils="11.1111" color="#cc1414" opacity="1" banded="0"/>
                    <connectors>`

With the change to `<wireExtras mils="11.1111"`

Which resulted in the following output in Fritzing when opening the file:
[image]

As seen above, it appears Frizting already supports altering the value of this variable and rendering it, so I just ask that it be user definable in the Inspector side menu when a wire is selected. Thanks.

cowboydaniel, 2022-06-11:
What about making the automatic PCB trace width to be user defined?

failiz, 2022-06-11:
That is the issue #3195

cowboydaniel, 2022-06-12:
Yes, but it should also be that way for auto routing also.

cowboydaniel, 2022-06-15:
[image]
[image]
I don't know if this is new or something, but the autorouter trace width IS setable. I just found this setting, and by selecting custom, you can change the default trace width!!!

KjellMorgenstern, 2022-06-15:
The graphical display is one thing. One way to implement it could be with a wire gauge property, with default settings according to https://en.wikipedia.org/wiki/IEC_60228 .

Most of those diameters would be unfit for breadboards, but might be useful for other illustrations or circuits.

[image]

szelenka, 2024-08-11:
This is possible in the PCB view in version 1.0.3:
- super fine (8 mil)
- extra thin (12 mil)
- thin (16 mil)
- standard (24 mil)
- thick (32 mil)
- extra thick (48 mil)

IMO, having it according to AWG would be the best to have in Breadboard view. 

Is there a PR open to expose the "mils" or "width" parameter in the Inspector for wires in Breadboard view? If not, can someone point me in the right direction for which section of the code needs to be altered to add this feature?