fritzing/fritzing-app #3595: Running qmake error while using libgit2-0.28.4.tar.gz

arimogi, 2020-01-20:
## Error while building Fritzing
I try to build Fritzing from the source. I have installed `boost` and `libgit2` as said in https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing and located those directories as siblings fritzing-app directory. Also, I have followed https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes.
While I ran `qmake` command, an error occurred as illustrated below:

[image]

It says:
`Project MESSAGE: found libgit2 include path at /path/to/fritzing-app/../libgit2/include`
`Project ERROR: static libgit2 library not found in /path/to/fritzing-app/../libgit2/build`

**Operating System:**
<Linux Ubuntu 18.04>

**Steps:**
- Installed `boost`
- Installed `libgit2`
- Created build in `libgit2`
- Built `libgit2`
- Execute `qmake` command

**Branch:**
develop branch

**Solution**
I tried to trace the `libgit2` directory and found no error. It is completely installed.
And then I opened the file `libgit2detect.pri` in `pri` directory inside `fritzing-app` folder, and I found this line:
` exists($$LIBGIT2LIB/libgit2.a) {`

On line _57_ that produced the error. The libgit2 build result has no "libgit2.a", so I create a symbolic link for libgit2.a from libgit2.so using this command
`ln -s libgit2.so libgit2.a`

After creating the symbolic link, I run qmake again, and it succeeds.

KjellMorgenstern, 2020-01-20:
That's amazing. Dynamic and static library files are usually not interchangeable. 
When building libgit2, run 
cmake -D BUILD_SHARED_LIBS=OFF
and then
cmake --build .
to get a .a file.

closed 2020-01-20 by KjellMorgenstern

haennes, 2021-09-17:
i got the Same Error too.
Solution:
 move the libgit2.a file into a subfolder called build

Nad42, 2022-07-16:
Hi all, would like to point out the issue with this code is there is a missing
"LIBGIT_STATIC = false"
on line 18/19 in libgit2detect.pri

soon as you add that program compiles fine if you have all dependencies installed.