fritzing/fritzing-app #3596: blog is out of date

skybrian, 2020-01-24:
As a new user, it took me a bit longer than it should have to figure out that Fritzing is not dead despite that article on Hackaday. One of the first things I looked at was the blog where the last post is from 2017, but that's misleading. Maybe someone could post something to the blog?

KjellMorgenstern, 2020-01-24:
You are absolutely right, thanks for the push. I will update it. Also, I will talk about Fritzing on FOSDEM in February.

closed 2020-01-24 by KjellMorgenstern

failiz, 2020-09-17:
Ping!
This has not be done. A lot of people just check the website and the blog and do not go to the github page. Thus, it still looks like the project has died. Look this recent post in the forum for an example:
https://forum.fritzing.org/t/paypal-payment-paywall-fritzing-status/10831
I think that some blog posts are really important, specially as there is a mandatory payment to download the program and the blog is the first thing that you see when you open the program. 

Some things to mention in the blogs: 
New maintainer
Release of the new version
Talk in FOSDEM
Roadmap
etc.