fritzing/fritzing-app #3599: OSX Check for MacPorts

meckiemac, 2020-01-30:
## Problem
I'm an active MacPorts (https://www.macports.org) user and installed (compiled) Fritzing successful on my Mac. Currently the build routine and build manual does not reflect this scenario. It requires some modification in the QMake project files. A typical MacPorts user is already in advanced mode and we would help him to develop its skills further if we would give them more support.

## Proposed Solution
A test for a MacPorts installation and small adjustments in the qmake scripts. I can help if wished.

It requires following changes:
All supporting libraries (boost & gitlib2) and tools(QT Creator) are provided by MacPorts and usually on the latest versions. Further due to a lack of static objects of gitlib2 a dynamic binding is required.
Currently I compiled Fritzing with Xcode or QT Creator (calling xcode/clang) directly. Depending on the compiler environment (via Xcode or via QT Creator) the modifications are a bit different. I go more the workaround way: fake the directories (symlinks) and extend include paths. General would be an adjustment of the library detection in the qmake scripts the best solution. 

Finally thank you for your great piece of software, it helped me to develop few IOT ideas further.

Kind regards,
Andreas

meckiemac, 2020-01-30:
Current changes:
Execute:
>sudo port install libgit2 boost qt5 qt5-qtcreator

Edit: phoenix.pro
* uncomment or remove: # load(configure)
* add in the 2nd macx { ... } part following (and adjust the paths if required):
   # additional Macports configuration options:
      boost_root = /opt/local/include
      QT_CONFIG -= no-pkg-config
      PKG_CONFIG = /opt/local/bin/pkg-config  
      LIBS += -L/opt/local/lib

* change "LIBGIT_STATIC = true" to "LIBGIT_STATIC = false"

KjellMorgenstern, 2020-01-30:
Could the dmg be adapted/used to support macports? Having two different compilation targets for Mojave seems undesirable. In the future (not yet scheduled), we wish to get an apple certificate.

meckiemac, 2020-01-30:
> Could the dmg be adapted/used to support macports? Having two different compilation targets for Mojave seems undesirable. In the future (not yet scheduled), we wish to get an apple certificate.

A good question I can't really answer since I never created anything for distribution. The good thing is that a local build will not get a complain from the OS. 
MacPorts covers with its umbrella all changes and misbehaves from the underlying OS and keeps the dependencies/requirements for the final app small. So I don't need to worry too much which OS version I have.

I can try by collecting the libraries and check if it runs somewhere else. That it how it looks like:
Fritzing:
	/opt/local/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1673.126.0)
	/System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 162.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/opt/local/lib/libiconv.2.dylib (compatibility version 9.0.0, current version 9.1.0)
	/opt/local/lib/libgit2.28.dylib (compatibility version 28.0.0, current version 0.28.3)
	/opt/local/libexec/qt5/lib/QtPrintSupport.framework/Versions/5/QtPrintSupport (compatibility version 5.14.0, current version 5.14.0)
	/opt/local/libexec/qt5/lib/QtSvg.framework/Versions/5/QtSvg (compatibility version 5.14.0, current version 5.14.0)
	/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.14.0, current version 5.14.0)
	/opt/local/libexec/qt5/lib/QtGui.framework/Versions/5/QtGui (compatibility version 5.14.0, current version 5.14.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1894.10.126)
	/System/Library/Frameworks/Metal.framework/Versions/A/Metal (compatibility version 1.0.0, current version 212.2.3)
	/opt/local/libexec/qt5/lib/QtConcurrent.framework/Versions/5/QtConcurrent (compatibility version 5.14.0, current version 5.14.0)
	/opt/local/libexec/qt5/lib/QtNetwork.framework/Versions/5/QtNetwork (compatibility version 5.14.0, current version 5.14.0)
	/opt/local/libexec/qt5/lib/QtSerialPort.framework/Versions/5/QtSerialPort (compatibility version 5.14.0, current version 5.14.0)
	/opt/local/libexec/qt5/lib/QtSql.framework/Versions/5/QtSql (compatibility version 5.14.0, current version 5.14.0)
	/opt/local/libexec/qt5/lib/QtXml.framework/Versions/5/QtXml (compatibility version 5.14.0, current version 5.14.0)
	/opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore (compatibility version 5.14.0, current version 5.14.0)
	/System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/AGL.framework/Versions/A/AGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 800.7.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)

KjellMorgenstern, 2020-01-30:
Although it might compile with Qt5.14, there are probably at least issues with rendering that users will run into. If those get reported, it might takes a lot of time to figure out that this is not from our official release.

meckiemac, 2020-01-30:
The QT version is less an issue, since the port system let me switch versions. Old version do not get  just simply uninstalled. As long I do not uninstall them I can de-activate the current version and activate the old version. I have no example currently for Qt since it's a quite new installation an no updates yet. But as example for perl:
> sudo port installed
...
perl5 @5.26.1_0+perl5_26 (active)
  perl5.26 @5.26.3_3
  perl5.26 @5.26.3_4 (active)
  perl5.28 @5.28.2_1
  perl5.28 @5.28.2_2 (active)
...

As an idea, since the MacPorts detection is roughly only 5-6 lines which does not influence the rest of the code, why not implement it with 2-3 lines documentation. The OSX self compile requires anyway few small further edits before it runs through, but the big work is done.