fritzing/fritzing-app #3600: Unreadable results in Search pane - white text on light gray

justinclift, 2020-01-31:
## Current Behaviour

Unfortunately, the results in the Search pane are unreadable.  White text on light gray seems an unfortunate choice.  And seems to be hard coded (?). :frowning: 

[image]

**Build:**
```
$ rpm -qa|grep -i fritzing
fritzing-parts-0.9.3b-lp151.2.1.noarch
fritzing-0.9.3b-lp151.3.5.x86_64
```
Ironically, guessing there's supposed to be text visible here (?):

[image]

**Operating System:**

SUSE OpenLeap 15.1 x64.

**Steps to reproduce:**
1. Start Fritzing.
2. In the "Search" pane, type in text to search for.  The problem is apparent here.

## Expected Behaviour

Results that are readable.  eg Aren't white text on light gray

justinclift, 2020-01-31:
Possibly related to #3336

justinclift, 2020-01-31:
As a workaround, setting my system desktop theme to be `Breeze` instead of `Breeze Dark` (then restarting Fritzing) allows the results to be readable.  eg black text on light gray.

If the background colour is hard coded to light gray, perhaps hard coding the text colour for the results pane would be the correct fix?

closed 2020-02-06 by KjellMorgenstern

justinclift, 2020-02-06:
Thanks @KjellMorgenstern. :smile:

Flavelius, 2023-05-29:
in 0.9.10 there still seem to be places where it doesn't match (white on grey):
[image]
Kubuntu 23.04, Plasma-style 'Wings Dark Plasma', application style 'Fluent round dark solid', color set 'Breeze Dracula'

commit e1891a21 (2020-02-06): Override system color for parts bin list. Fixes #3600