fritzing/fritzing-app #3601: Can't extend leg in breadboard if connected

daxliniere, 2020-02-01:
## Current Behaviour
Create a circuit in schematic view, switch to breadboard view = all components will have dotted lines indicating connections.
Bug: You cannot extend the leg on a component if it is connected in schematic. I was only able to extend component legs if I placed a fresh, unconnected component on the board. Then it worked perfectly.

Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
Windows 10

**Steps to reproduce:**
See above.

## Expected Behaviour
See above.

daxliniere, 2020-02-01:
UPDATE: I just discovered I was able to bend the leg of a 5mm red resistor, but when I did so, that leg became disconnected.

KjellMorgenstern, 2020-02-03:
In the  Breadboard view:
This is a fast way to add wires in breadboard view, as such it is a feature.
Extending the leg is still available, but hard to reach.
Usually the wire is in front of the leg, and gets picked up first. There is a small area on the leg end, that is not overlapped by the wire (zoom in if necessary). 

If you select it there, on the legs endpoint, as far as possible from the wires endpoin, you can extend the leg.
However it is then also disconnected. This is in line with what happens when you select the end of an already connected wire. 

Treating connections as a bendpoint instead of two connected endpoints might be an idea to get a more intuitive behavoir.

Improving the possibility to select the right one of multiple overlapping areas is a general issues, that affects all interactions, not only extending legs.

I could not reproduce the disconnection when bending the leg.

daxliniere, 2020-02-03:
Hi Kjell,
I have designed a circuit in schematic view. Now that I'm doing my stripboard layout, I have decided/discovered that I need some LEDs to span 3 tracks instead of 2. Currently, I cannot find a way to do this that doesn't require me to fix a lost connection.
Surely this would not be considered a feature?

[image]

mMerlin, 2020-02-06:
Ignoring the `feature` of disconnecting when bending, a technique to help select the bendable leg, is to temporarily turn off the wire and/or ratsnest layers in the breadboard view.  With them out of the way, the leg end is much easier to select.

`View ¦ Wire Layer` and `View ¦ Ratsnest Layer`

The same is true in other views, when you are (say) trying to manipulate an existing wire, and keep getting a ratsnest line instead.  Just turn off the layers you do not currently need to simplify the view.

KjellMorgenstern, 2020-02-06:
@daxliniere  If disconnecting it was what you intended, how would you do it?
We could code it, so that in this very specific situation, the behavior differs, and the leg will be extended while keeping the connection.
This would however not make sense if the connection was to a breadboard, for example. So it would have to be done carefully, and under consideration of probably some other sidecases. 

@mMerlin Thanks for this hint. I am thinking about adding a feature to use the mouse wheel to help with quickly selecting layers / overlapping components.