fritzing/fritzing-app #3602: Schematic view - snap "magnets" aren't strong enough

daxliniere, 2020-02-01:
## Current Behaviour
Could a dev please adjust the snap threshold in the Schematic view? I find that when I create a new point on a wire, even when the wire was perfectly aligned to the grid, that point is always created SLIGHTLY off the grid and needs to be moved on. Even moving it on isn't easy every time.
It seems like the virtual 'magnets' or 'detents' aren't strong enough to pull the points to the grid intersections.

(Separately, thank you for your amazing work. I'm glad to be a financial contributor and really do appreciate this project. :) )

**Build:**
Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
Windows 10

KjellMorgenstern, 2020-02-03:
Which component did you use? Maybe there is an issue with parts that define a very small endpoint. Connecting wires to a resistor in schematics view is a pretty standard task, and I don't find it difficult at all (but I am biased :-) )

daxliniere, 2020-02-03:
Hey Kjell! Thanks for taking the time to answer my reports.
I'm referring to the wires in Schematic view. Please see the attached screencap.
[image]

failiz, 2021-11-02:
Duplicate of #3738

closed 2022-06-01 by KjellMorgenstern