fritzing/fritzing-app #3604: FR: Replace component

daxliniere, 2020-02-01:
If a component has the same pinout and size, you should be able to replace that component with another. This would preserve all wiring and save some time.

For example, replacing a rectifier diode with a zener diode.

Thank you for reading. :)

KjellMorgenstern, 2020-02-07:
Use delete minus to remove the rectifier diode, add the zener diode and reconnect the loose wires.

In case of the rectifier diode, you can also simply modify the diode to be a zender diode via the Inspector menu.

closed 2020-02-07 by KjellMorgenstern

KjellMorgenstern, 2020-02-07:
[image]

daxliniere, 2020-02-07:
"...and reconnect the loose wires."
That's the bit I didn't want to have to do. I'm thinking about 16 pin ICs, etc, but from your animation, it looks like it'll automatically connect. I haven't seen that behaviour so far. Does it work in the schematic view? Perhaps that's my real feature request, then.