fritzing/fritzing-app #3608: Incorrect wires added during breadboard layout

daxliniere, 2020-02-01:
## Current Behaviour
In the Breadboard view, on a stripboard, if you place a component that creates an unwanted connection, re-positioning that component doesn't remove the unwanted connection. That unwanted connection becomes a permanent feature and the only solution is to undo many, many steps or to delete 4-5 parts and re-do the schematic and breadboard layout.

I have even tried to convert the unwanted connection into a wire in the schematic view and then delete that wire, but the unwanted connection is left behind.

**Build:**
Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
Windows 10

**Steps to reproduce:**
See above.

## Expected Behaviour
See above.

KjellMorgenstern, 2020-02-01:
Can you add a screenshot? On a stripboard, it should be possible to simply cut connections.

daxliniere, 2020-02-02:
Hi Kjell,
I have captured a series of 'states' which you can see below. Simply copy each one and go into Breadboard view and paste them.

daxliniere, 2020-02-02:
Everything good:

<module fritzingVersion="0.9.4.2019-12-01.CD-498-0-a1ffcea">
 <boundingRects>
  <boundingRect rect="0 0 0 0" name="PCB View"/>
  <boundingRect rect="-45 -456.96 513.774 457.21" name="Breadboard View"/>
  <boundingRect rect="0 0 0 0" name="Schematic View"/>
 </boundingRects>
 <instances>
  <instance path=":/resources/parts/core/30.20Stripboard2ModuleID.fzp" moduleIdRef="30.20Stripboard2ModuleID" modelIndex="85413388">
   <property name="size" value="30.20"/>
   <property name="buses" value="20.1h 28.12h 16.0h 21.5h 18.8h 13.2h 11.17h 7.7h 18.6h 2.6h 12.2h 22.3h 22.17h 26.13h 24.10h 9.2h 24.12h 17.15h 21.15h 21.9h 26.2h 27.13h 12.13h 4.10h 15.2h 5.4h 24.9h 11.1h 25.0h 19.8h 13.9h 27.17h 12.14h 26.17h 14.8h 1.17h 4.3h 3.2h 16.19h 4.6h 13.7h 2.4h 9.16h 5.14h 24.2h 10.6h 7.14h 7.10h 3.8h 26.5h 15.8h 20.15h 11.12h 28.4h 16.11h 18.0h 26.1h 14.18h 21.11h 21.11v 1.9h 21.3h 4.8h 25.11h 8.11h 4.1h 15.6h 4.18h 12.11h 23.19h 1.4h 6.10h 25.8h 10.1h 27.12h 6.14h 6.0h 11.9h 22.10h 28.2h 19.13h 18.4h 1.2h 3.7h 3.19h 16.7h 22.18h 16.8h 27.15h 3.3h 18.10h 9.7h 24.0h 14.4h 14.16h 1.5h 19.3h 16.1h 24.4h 22.2h 18.9h 20.11h 1.3h 16.4h 14.5h 16.9h 11.13h 1.16h 5.3h 9.3h 14.19h 23.15h 2.16h 10.2h 5.18h 0.16h 27.16h 6.13h 25.15h 26.15h 9.15h 3.6h 0.3h 0.19h 18.3h 26.7h 5.19h 24.3h 27.10h 1.10h 0.0h 1.19h 21.14h 12.18h 1.12h 7.6h 6.11h 19.14h 22.11h 0.13h 0.15h 6.18h 1.18h 19.5h 0.17h 21.10h 21.7h 21.7v 21.13h 1.1h 25.19h 0.12h 13.14h 15.4h 8.15h 25.13h 2.17h 6.15h 24.17h 10.9h 4.9h 18.15h 7.4h 9.11h 28.10h 5.17h 15.0h 27.5h 8.16h 10.7h 15.12h 20.13h 25.12h 10.11h 22.12h 17.6h 19.17h 4.4h 28.3h 5.9h 20.16h 19.1h 26.8h 13.4h 8.5h 16.5h 15.7h 22.0h 27.18h 22.16h 12.16h 23.17h 19.11h 16.6h 9.19h 5.13h 28.11h 15.13h 7.0h 12.15h 12.8h 12.3h 21.1h 16.10h 9.9h 15.19h 2.0h 25.9h 11.8h 27.8h 20.18h 23.3h 19.18h 6.3h 9.13h 24.7h 26.0h 23.13h 2.19h 0.14h 0.2h 15.9h 3.12h 2.3h 2.8h 26.6h 27.1h 27.6h 10.3h 20.19h 3.17h 16.16h 6.12h 5.1h 9.8h 4.13h 11.0h 26.16h 28.19h 4.11h 11.5h 19.2h 7.5h 4.7h 4.7v 25.7h 25.3h 7.16h 9.12h 23.16h 8.7h 8.7v 8.17h 8.6h 24.6h 6.7h 3.10h 5.7h 15.1h 7.13h 5.2h 14.0h 9.14h 28.18h 13.6h 10.19h 20.3h 0.4h 5.5h 11.15h 3.5h 6.8h 5.10h 18.13h 25.5h 26.12h 0.7h 6.19h 16.15h 3.4h 22.7h 11.2h 1.11h 12.12h 26.14h 15.3h 10.15h 14.11h 11.11h 1.6h 20.8h 28.8h 10.12h 10.17h 8.9h 23.12h 17.1h 17.0h 20.7h 4.2h 15.11h 4.17h 24.8h 26.4h 18.17h 24.5h 6.9h 18.12h 27.4h 7.15h 19.12h 13.8h 19.9h 4.14h 2.5h 12.5h 7.8h 12.1h 17.7h 17.7v 14.2h 23.14h 15.10h 14.12h 15.18h 9.10h 21.12h 14.17h 17.5h 7.19h 24.18h 24.15h 9.4h 13.3h 7.18h 8.3h 17.19h 8.4h 4.19h 6.5h 13.16h 18.19h 1.7h 2.13h 23.7h 2.15h 20.2h 11.10h 3.14h 5.0h 5.15h 14.6h 7.11h 18.11h 22.14h 28.6h 11.18h 22.15h 22.6h 2.14h 21.2h 2.9h 16.12h 18.2h 28.9h 14.1h 28.1h 12.6h 25.16h 21.18h 24.16h 14.15h 12.10h 17.18h 23.9h 19.19h 14.7h 2.7h 23.4h 19.10h 22.8h 27.9h 0.8h 3.9h 11.6h 26.11h 17.17h 13.18h 8.10h 17.13h 8.0h 25.1h 12.9h 17.8h 13.11h 17.2h 16.17h 25.2h 15.15h 27.3h 13.1h 9.18h 20.5h 17.4h 0.5h 14.3h 3.0h 17.10h 23.2h 2.10h 23.5h 21.0h 24.13h 18.16h 26.18h 8.14h 6.17h 4.12h 19.16h 23.8h 3.16h 18.7h 10.10h 11.14h 0.10h 9.0h 23.11h 17.16h 5.6h 2.12h 1.8h 18.5h 23.1h 27.11h 24.11h 25.17h 0.6h 25.4h 12.19h 10.5h 20.6h 2.11h 13.0h 3.1h 15.14h 21.4h 24.19h 4.15h 7.1h 2.2h 20.0h 28.17h 8.2h 12.4h 20.17h 25.6h 0.9h 0.11h 11.16h 6.16h 5.16h 19.4h 19.7h 23.0h 1.14h 22.13h 19.6h 0.18h 16.2h 28.14h 6.1h 21.6h 0.1h 17.12h 22.5h 27.14h 13.5h 17.11h 28.5h 12.7h 9.1h 6.2h 14.13h 21.19h 23.10h 16.3h 20.4h 16.14h 4.16h 13.10h 18.1h 10.13h 14.10h 13.12h 11.4h 18.14h 10.18h 20.10h 8.12h 16.18h 13.17h 12.0h 10.0h 13.19h 3.15h 14.9h 3.11h 28.13h 25.10h 23.18h 5.8h 10.4h 17.9h 26.19h 28.16h 1.0h 21.17h 20.9h 5.12h 15.16h 20.12h 8.19h 4.0h 22.9h 8.18h 4.5h 7.9h 25.18h 25.14h 11.19h 16.13h 8.1h 1.15h 24.1h 22.4h 21.16h 27.19h 27.2h 2.1h 7.2h 18.18h 3.13h 1.13h 26.10h 7.12h 17.3h 17.14h 9.5h 10.8h 26.9h 26.3h 28.0h 27.7h 15.17h 14.14h 19.0h 24.14h 28.15h 7.3h 2.18h 9.17h 12.17h 7.17h 9.6h 13.13h 10.16h 8.8h 20.14h 6.6h 23.6h 28.7h 6.4h 27.0h 15.5h 10.14h 21.8h 22.19h 19.15h 13.15h 11.3h 11.7h 22.1h 5.11h 3.18h 8.13h "/>
   <property name="layout" value="vertical strips"/>
   <title>Stripboard1</title>
   <views>
    <breadboardView layer="breadboardbreadboard">
     <geometry y="-315" z="1.50002" x="-45"/>
     <connectors>
      <connector connectorId="connector5021" layer="breadboardbreadboard">
       <geometry y="0" x="0"/>
       <connects>
        <connect connectorId="connector1" modelIndex="85424605" layer="breadboard"/>
       </connects>
      </connector>
      <connector connectorId="connector10004" layer="breadboardbreadboard">
       <geometry y="0" x="0"/>
       <connects>
        <connect connectorId="connector1" modelIndex="85424609" layer="breadboard"/>
       </connects>
      </connector>
      <connector connectorId="connector6004" layer="breadboardbreadboard">
       <geometry y="0" x="0"/>
       <connects>
        <connect connectorId="connector0" modelIndex="85424609" layer="breadboard"/>
       </connects>
      </connector>
      <connector connectorId="connector18" layer="breadboardbreadboard">
       <geometry y="0" x="0"/>
       <connects>
        <connect connectorId="connector1" modelIndex="85430351" layer="breadboardWire"/>
       </connects>
      </connector>
      <connector connectorId="connector11021" layer="breadboardbreadboard">
       <geometry y="0" x="0"/>
       <connects>
[... cut ...]
       <geometry y="0" x="0"/>
       <connects>
        <connect connectorId="connector16012" modelIndex="85413388" layer="breadboardbreadboard"/>
       </connects>
      </connector>
      <connector connectorId="connector1" layer="copper1trace">
       <geometry y="0" x="0"/>
       <connects>
        <connect connectorId="connector19012" modelIndex="85413388" layer="breadboardbreadboard"/>
       </connects>
      </connector>
     </connectors>
    </pcbView>
   </views>
  </instance>
 </instances>
</module>

KjellMorgenstern, 2020-02-03:
[image]

I loaded Fritzing - fault 1.fz
Could you mark where the problem is?

KjellMorgenstern, 2020-02-03:
PS: Later on there are issues, which seem to be from resizing the board, looks like #3162 .

There is another issue about unwanted connections, maybe this is related? #3130

daxliniere, 2020-02-03:
Hey Kjell, to answer your first question, the schematic view will show a problem. A dotted connection has been made where it shouldn't be.
(No idea why, but I can't seem to copy/paste the contents of those TXT files all of a sudden. ????)

daxliniere, 2020-02-03:
I'm not sure where to post this or even what to call it, but I had another issue. Once you've started editing in the Breadboard page, things start to get wacky in the schematic view, like there's no designated 'master'. I feel like the schematic view needs to have 'master status' over all other pages, since that is the core of every circuit.

For example, I just came across this problem: I wanted to disconnect the limit switches at the top of the schematic so I could connect them via a screw terminal block. As you'll see in the screencap, when I disconnected one of the wires, it was automatically 'replaced'. This is 'clever', but absolutely not wanted behaviour.

[image]

daxliniere, 2020-02-04:
CAUGHT ON CAMERA!
Also attached are before and after states.

[image]

[Bug - before and after.zip](https://github.com/fritzing/fritzing-app/files/4150882/Bug.-.before.and.after.zip)

daxliniere, 2020-02-04:
Sorry guys, but this is completely unusable. Look how the transistor pins are joined incorrectly and for absolutely no reason:

[image]

daxliniere, 2020-02-04:
Also, shouldn't the colour of the dotted line on the Breadboard view match the Schematic view? It would make it a whole lot easier to tell where a short had occurred.

Thanks again, Kjell!

mMerlin, 2020-02-06:
Deleting a wire should convert to a ratsnest line **only** if there is a wire/trace on another view for that connection.  You can also delete that ratsnest line as a separate step, but beware of what connections might be in place on other views.  It's not hard to get into crazy states across different views.  I suggest finding the wire in the other view(s), and deleting them before deleting the ratsnest lines.

vanepp, 2020-02-06:
It is best practice to make all your connections in one view (breadboard, schematic or pcb.) Once that is completed switch to the other views where rats nest lines will indicate the connections set in the completed view, and click on the rats nest line to create a wire and drag it to route it correctly. If you make a conflicting connection in another view, it is possible to corrupt the routing database (which may be a bug, or just how it works) in such a way that you can not recover. In that case the only thing  you can do is start the sketch again.  As to your text files above, you need to add a .zip to the end of the .fzz file as in Fritzing - everything good.fzz.zip so github will upload it. It looks to have only loaded the .fz file and not the associated svg files as a .txt

KjellMorgenstern, 2020-02-06:
Deleting a wire *or* ratsnest line in breadboard or schematics view should not be possible if it is already routed on the PCB view. And the other way round, too. This avoids deleting a already routed connection by accident. It might make sense to add a hint if users try to delete such a connection. Please open a feature request for this if you agree.

Regarding the unwanted connection: I stil could not reproduce it from the first examples, but from the video it seems that a connection is created while hovering the part over the stripboard, before even setting it down. That indeed is a bug that should be fixed, as it makes striboards unusable/very difficult to use should it occur.
I think it is the bug we should focus on in this issue.

vanepp, 2020-02-06:
"but from the video it seems that a connection is created while hovering the part over the stripboard, before even setting it down." 

I am unsure that is what is happening here. It looks to me from the pictures that the routing database is corrupted by incorrect connections (getting complete copies of the fzz files would probably confirm that.) If so then the unwanted connections are caused by the routing corruption and will likely go away when it is fixed. I am unsure whether the corruption is a bug or an unavoidable consequence of an incorrect connection in another view causing a change that can't be reversed. I'm leaning towards a bug because I have an example where removing all connections in all views does not correct the routing database (from an example posted lately in the fritzing forums) which I think is incorrect.

mMerlin, 2020-02-06:
@KjellMorgenstern Deleting a ratsnest line in one view when there is a matching routed connection in another view should not be possible for reasons mentioned.  However, deleting a wire, which then reverts to a ratsnest line **should** be possible.  If editing/routing the wire/trace gets messed up bad enough, it is a lot easier to just delete it, and start that particular wire/trace over again.  Especially when overlapping wire/parts make selection of the pesky extra bend point all but impossible.  Even with simplification by reducing the number of displayed layers.  Sometimes, move to front/back can help, but often delete and recreate more carefully is easier.

KjellMorgenstern, 2020-02-07:
In order to resolve this issue I must find a way to reproduce it. 

I agree with @vanepp that uploading the fzp of a faulty project (and annotating again what is the fault)  is worth a try.

Otherwise I will have to close this issue unresolved, as there are just to many use cases and uncertainties, maybe even multiple different issues, mixed into this. Chances of isolating the issue and thus fixing it seem dim.

KjellMorgenstern, 2020-02-07:
Update: I overlooked the file "Bug - before and after.zip" . With that file I could now reproduce the issue of creating a short over the resistors.

KjellMorgenstern, 2020-02-07:
In regards of deleting the ratsnest line over R2: Right click and choose "delete ratsnest line" worked for me, both in Breadboard and schematics view. 
Limiting the displayed layers via Menu -> View might make sense, too, depending on your style of working.

vanepp, 2020-02-07:
I agree this isn't a bug but operator error (having also overlooked "Bug - before and after.zip".) You have correctly completed schematic before starting on breadboard, now however your best bet is to click and drag on a rats nest line in breadboard which will create a wire to where it should connect. In the case of D1 that is one row down from where it is connected in breadboard to connect to the strip that is shorted to the terminal block connection As well I would change the short between the two rows at the screw connector to a wire like this to make the connection more obvious. as it will be wired in real life (note the resistor had to move one position left to allow for the connection of the diode as well):

[image]

Your problem here is that breadboard connections reflect back in to schematic, so an incorrect connection in breadboard will create rats nest lines in schematic and if you then click on the new rats nest line in schematic, in some cases corrupt the routing database in such a way that the only solution is to restart by clearing all connections. I don't think there is a bug here. I would also make a cut in the strip board just after the anode of D1 to avoid making an unwanted connection to the rest of that strip elsewhere on the board that I didn't do here. If you see an unexpected rats nest line, check the other 2 views for an error.

KjellMorgenstern, 2020-02-07:
@vanepp "...and if you then click on the new rats nest line in schematic, in some cases corrupt the routing database in such a way that the only solution is to restart by clearing all connections."

Do you have an example for this? Database corruption would indeed be a bug, but I did not yet see it happening here. Or did you mean #3580 ?

vanepp, 2020-02-07:
@KjellMorgenstern  Unfortunately not that I can reproducible dependably. The fzz in #3615 initially showed this behavior, deleting all wires in all views then redoing breadboard initially added spurious rats nest lines which is the sign of data base corruption. Now however starting from the original fzz (and not making the error in 3615, although I don't think that is involved) I can't get the spurious rats nest line to appear again. There have been a handful of cases of this over the yeas in the forums and our usual advise has been to start fresh and don' repeat the mistake. I have never been sure if the cause is a bug, or the fact that incorrectly shorting two nets is not reversible, it may be either. I was hoping to start from the fzz in 3615, delete all the errors in schematic and then re introduce them to be able to reproduce the failure from the start so we can see what happens, but so far haven't even been able to reproduce the problem on the original file except for the first time (which may have something to do with the order that I did things in or I may have missed a wire.) I'll keep poking, because if it is a bug it would be a good one to fix, and knowing how it occurs we may find another cure.

daxliniere, 2020-02-07:
Hey Vanepp, thank you for looking into this as well.

> It is best practice to make all your connections in one view (breadboard, schematic or pcb.) 

Yes, I'm aware of this and that's exactly what I was trying to do. In fact, I think it should be possible to 'lock' a circuit from being able to be modified in a secondary view. (This is essentially being able to set a 'master editor view' as I suggested earlier).

daxliniere, 2020-02-07:
@vanepp that bridged track was well-spotted. I have to admit, that wasn't a connection I had intended to make. I have found it all too easy to accidentally make bridged connections on the stripboard.
Is there a way we can change that UI? Perhaps you have to CTRL+click in order to bridge and cut the strips? That would help a lot.

But yes, I had situations where the corruption happened, I would undo several steps, move one of the components sitting off the stripboard and try again and it worked.

daxliniere, 2020-02-08:
At the very least, can we get an alternate stripboard layout with high contrast colours, please? Flat, non-shiny copper on phenolic board doesn't really have much contrast.

vanepp, 2020-02-08:
@daxliniere "In fact, I think it should be possible to 'lock' a circuit from being able to be modified in a secondary view. (This is essentially being able to set a 'master editor view' as I suggested earlier)." 

Unfortunately that isn't likely to be easy to do. The code is set up to reflect changes in one view in to the other two views, and selectively stopping that may not be easy. It may be possible to cause a dialog box to appear if a change in a view is different than a current rats nest line, requiring confirmation that the change is intended but I'm not sure how easy that would be either. The code is complex and changes may have unanticipated results.

 "Is there a way we can change that UI? Perhaps you have to CTRL+click in order to bridge and cut the strips? That would help a lot." 

Yes almost anything can be done. The problem has been finding someone capable and willing to do the work. Development has stopped for so long that no one is even making pull requests any more (I think around a dozen got included in the 0.9.4 release over the 4 years from the last release. The hope is with increased donations to be able to hire professional developers to move forwards (the code base is too complex for amateurs to deal with easily.) To that end you should likely file enhancement requests for the features you want added. No guarantee they will make it in, but they can be up voted by folks here (al;though I'm not clear on how you do that) to indicate interest. If there is a lot of interest in an enhancement it presumably will get priority. 

"At the very least, can we get an alternate stripboard layout with high contrast colours, please? Flat, non-shiny copper on phenolic board doesn't really have much contrast."

While again an enhancement, this one should be relatively easy, just change the colors in the source (or better yet, make them user settable from the UI, more work but more flexible too.)

" that bridged track was well-spotted. I have to admit, that wasn't a connection I had intended to make. I have found it all too easy to accidentally make bridged connections on the stripboard."

That one was fairly east to find. All I did was drag D1 off the strip board like this:

[image]

and then look where the rats nest lines go. In this case that indicated the diode was connected to the wrong end of the resistor (causing a short in schematic.)  I had assumed the jumper was intentional (although in the real world would need to be a wire as perf board won't do the jumper, because the diode won't fit directly between the resistor and the strip it wants to connect to and you thus need to wire to another strip to provide the needed distance, or use a top view diode which will fit in a .1in pair of holes (I think I or someone else made one some time back, it should be in the forums).  Probably again a change to the UI to allow the user to select whether or not the jumper can be done is probably the answer (I think the intention for the jumper is to create custom perf board layouts, rather than to serve as a jumper.)  There are likely to be people who want the current method to still work and making the new behavior user setable with the current the default is probably the best way forward. As well you can click on a strip (or any connection) and it will light up everything connected to that in yellow which would also have indicated that connection.

KjellMorgenstern, 2020-02-08:
I think the original issue is resolved (better: explained) as good as possible. I'd suggest to close this issue. If you want, please open new issues for the findings made during this discussion.

daxliniere, 2020-02-09:
I thought you guys WERE the developers. You mean there's no-one at the wheel?

closed 2020-02-09 by KjellMorgenstern