fritzing/fritzing-app #3609: Bendpoints are not moved together with the selection.

KjellMorgenstern, 2020-02-03:
## Current Behaviour
"In any view, when box-selecting (say) the left half of my circuit, when I drag-move the selection, most of the wire bend points stay where they are, resulting in a big mess of bend points that I have to manually straighten out. It seems like the components themselves, and any wires directly between them, are moved OK, but unconnected wire segments can't be moved on their own."

Bendpoints of unconnected wires and bendpoints of a wire partly selected are not moved together with the selection.

**Build:**
Version 0.9.5 
(dCD-380-66-g29b8f5d3 2019-10-24) 64 [Qt 5.9.5]

**Operating System:**
Ubuntu 18.04

**Steps to reproduce:**
- Add some elements, eg. resistors
- Connect them, use bendpoints
- Select elements and some bendpoints, drag the selection

## Expected Behaviour
Bendpoints should move together with the selection.

KjellMorgenstern, 2020-02-09:
[image]

Expected:
The bendpoint to the bottom right should be part of the selection.

JC3, 2021-11-25:
Related: https://forum.fritzing.org/t/move-group-of-parts-wires-in-pcb-view/14451