fritzing/fritzing-app #3610: setting property "editable pin labels" to "true" results in incorrect representation of schematic graphic

Notedop, 2020-02-03:
## Current Behaviour

When creating a new part and setting the property "editable pin labels" to 'true', the representation of the schematic part is reset to an IC part instead of the loaded svg. Part attached at bottom 

**Schematic view of part in parts editor**
[image]

**View of part in schematic, with 'Editable pin labels' set to 'false' (original value)**

[image]

**View of part in schematic view, with 'Editable pin labels' set to 'true' (original value)** 

[image]

**Build:**
Version 0.9.4 (CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]
Seems to be existing older issue as I could find someone else stating the same [here ](http://fritzing.org/forum/thread/4164/)

**Operating System:**
Windows 10

**Steps to reproduce:**

- Select 8 pins IC from the Core and go into parts editor
- Load breadboard, schematic, pcb and icon svg
- Link the pins in breadboard, schematic and pcb to their respecive graphic
- Update the metadata --> properties where it states "editable pin labels" and set it's value to 'true'
- Save the part as a new part (i've used prefix JST)
- Use the part in your design and verify the Schematic view.

**Result**

1. Editable Pins button is enabled and pin names are editable via the Schematic view
2. Graphic has changed to a different graphic than loaded into the part during part creation. It seems to be the default 8 pin IC graphic 
3. Pins cannot be linked, atleast for this particular part only 2 of 8 pins are 'linkable' in the schema view.
4. When editing the part again, it looks like that the "editable pin labels" has been reset to false again. When Saving the part once more, the part is showing the loaded graphic however the editable pins functionality is disabled.

## Expected Behaviour
1. Correct behavior
2. Graphic loaded during part creation for the schematic should be visible
3. Pin to graphic mapping should persist
4. "editable pin labels" property should be visible as 'true' as this is how we saved the part.

Note: It would be good to see a full overview of which properties can be set and are used to enable specific functionalities for parts usage. It might be that setting this property, requires more actions for it to work with a part; if that is so, then it needs to be made clear to the user.

[JST-XH.zip](https://github.com/fritzing/fritzing-app/files/4149888/JST-XH.zip)

vanepp, 2020-02-06:
It appears editable pin labels only works for dip packages. Manually setting the property in the fzp file (without using parts editor at all) creates the same problem. It appears the code swaps in a 8pin dip package in schematic if editable pin labels is set (and the connectors are then labeled wrong somehow which is what causes the red square.) As a side note your original schematic svg is lacking terminalIds which will cause the wires to connect at the center of the pin rather than the bottom end as they should.

Notedop, 2020-02-08:
> As a side note your original schematic svg is lacking terminalIds which will cause the wires to connect at the center of the pin rather than the bottom end as they should.

Thank you for the feedback. So for terminalId's I would create another rectangle on top of each pin in the SVG and give it a ID ending with 'terminal' ? Is that how Fritzing automatically identifies and links the pin to the terminal or is there some other actions required to be taken within the parts-editor.

> The terminalId attribute is optional. Each connector takes up a certain area in a part (in the example above, it's a rectangle), but a wire will actually attach to a connector at a single point within the connector's area. This point of attachment is called a terminal point. The default terminal point is the center of the connector area. If you want the terminal point elsewhere, the terminalId attribute points to yet another element in the SVG file. 

`<p svgId="connector1pin" layer="schematic" terminalId="connector1terminal" />`

Btw, I will be able to do testing for this and the other bug i've logged once a fix is available. Just need to know which branch to checkout ;-)

vanepp, 2020-02-08:
@Notedop "Thank you for the feedback. So for terminalId's I would create another rectangle on top of each pin in the SVG and give it a ID ending with 'terminal' ? Is that how Fritzing automatically identifies and links the pin to the terminal or is there some other actions required to be taken within the parts-editor."

There are two options: you can edit the part with parts editor and select a connector then in the terminal point area of the right window select center (what you have now), n, s, e, w, which will create a terminal id at the selected compass point of the connector pin. I have never been able to get this to work correctly and thus don't use parts editor. The way I do it is to create a 10 thou by 10 thou invisible rectangle with svg xml that looks like this:

    <rect
      id="connector4terminal"
      x="0.16988206"
      width="10"
      height="10"
      y="500.27841"
      fill="none"
      stroke-width="0"
    />

which equates to a style like this in xml editor in Inkscape: style fill:none;stroke-width:0
(you will find some of the parts factory generate terminalIds with 0 width, which I consider a bug that I need to fix, as Inkscape will not select a 0 width element making them hard to move.) Then move this rectangle to the appropriate place on the terminal pin.  As to the original problem, it appears there is a hard coded svg which gets substituted for the schematic svg if editable pin labels is true. This overrides the schematic svg (that just bit me on a part I made for someone,) I consider this one a bug that needs fixing too. There are tutorials in the Fritzing forums on how I make parts that may help you with some of this although it assumes you are comfortable with editing xml directly (a number of people think that parts editor is the better way to go, but I disagree.)

https://forum.fritzing.org/t/part-creation-howto-part-1-breadboard-and-pcb/7692

deshipu, 2022-09-14:
I have stumbled upon a bug related to this. I don't mind that Fritzing generates a DIP-like schematic symbol for the part, but I do mind that there is an off-by-one error with pin labeling an numbering, and that the missing pin is assigned to the symbol as a whole - this is why it is red on the screenshot.

[image]

vanepp, 2022-09-16:
@deshipu : You should probably open a new issue (with how to reproduce the problem) as it may or may not be related to this issue.

deshipu, 2022-09-16:
Thanks, I will do that.

closed 2023-08-13 by KjellMorgenstern