fritzing/fritzing-app #3611: SVG's of parts are not cleared/deleted

Notedop, 2020-02-03:
## Current Behavior

When editing/creating parts and loading new SVG's onto a specific part, the old SVG is not removed from the user parts directory, resulting in unneeded overhead of SVG files.

These svg files are result of trying to create a single part. I had to load the graphics multiple times while creating my part (first time). Effectively only one of the selected svg's is used, the remainder is overhead.

[image]

**Build:**
Version 0.9.4 (CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
Windows 10

**Steps to reproduce:**

1. Select 8 pins IC from core and open Parts Editor
2. In any of the views; File --> Load image for View. In this example we'll use Breadboard view only.
3. Select a svg you want to load for the specific view
4. Save the part (as a new part). Use a prefix which can be easily identified. I used "TOO_MANY_FILES"
5. Repeat step 2,3,4 a few times. I've done it 2 times in total
5. Verify the content of directory: C:\Users\*yourusernamehere*\Documents\Fritzing\parts\svg\user\breadboard
6. Result is that multiple SVG's are created. In this example i have only iterated step 2-4 only two times, which resulted in 3 SVG files. The resulting svg files only seem to differ in some whitespace.

It's strange that with 2 iterations, 3 svg's have been created. 

[image]

Breadboard SVG files attached
[breadboard.zip](https://github.com/fritzing/fritzing-app/files/4150123/breadboard.zip)

Part
[TO_MANY_FILES__b34bef3ea5dcdf356c5afd14904a7293_2.zip](https://github.com/fritzing/fritzing-app/files/4150128/TO_MANY_FILES__b34bef3ea5dcdf356c5afd14904a7293_2.zip)

## Expected Behavior

When loading a graphic for a view (breadboard, schematic, pcb, icon), the old graphic should be removed. The moment when this is done, can differ. For example one could do it only after saving as one does not know if a new part should be created, keeping the original part intact, or if the existing part should be overwritten.

Notedop, 2020-02-03:
Maybe better expected behavior
When saving a custom user part and the graphics have been updated, the old graphic should be cleaned up.

Although this is dependent on the workflow within the application and how things work 'under the hood'

KjellMorgenstern, 2020-02-04:
Might be related to commit ae2817577ec , where a similar issue was fixed.

vanepp, 2020-02-06:
"Might be related to commit ae28175 , where a similar issue was fixed." 

It doesn't look like it. Before the commit parts editor did not register the svg changes with the main window thread, so if you exit Fritzing and choose to not save the part it didn't get deleted (with the result you couldn't reload the part without clearing the user directories). This looks to be a case where it isn't deleting a previously loaded svg when a new version is loaded which won't break Fritzing (which will use the new svg name) but will clutter the file system. I think a similar thing is occurring in Appdata/Local/tmp, as I recall Fritizing creates tmp files and expects Qt to remove them but it doesn't always happen (or perhaps never happens) and they stay there forever unless you have a tmp clearing program (at least on Windows, I suspect they are in /tmp on Unix and thus get cleared on boot.)