fritzing/fritzing-app #3612: FR: Disconnect part

daxliniere, 2020-02-03:
## Problem
I want to remove a part from circuit without deleting it. Currently we have to select each wire one by one and delete those wires.

## Proposed Solution
Add an option to the context menu for "Disconnect part".
This would be found alongside Rotate, Flip, Copy, Duplicate, Delete, Delete (minus), etc.

vanepp, 2020-02-06:
There already is one. Delete the part via "delete minus" rather than "delete" and it will leave the traces in place while deleting the part. Once you replace the part you need to drag the wire on each connection to remake the connection (should change from red to green to indicate the connection has established.)

KjellMorgenstern, 2020-02-07:
I also understand that "delete minus" is what was asked for.

closed 2020-02-07 by KjellMorgenstern