fritzing/fritzing-app #3615: Incomplete routing on breadboard connection

vanepp, 2020-02-06:
## Current Behaviour
Text

[Warbringer-new-bb-routed-routing error-minimal-case.fzz.zip](https://github.com/fritzing/fritzing-app/files/4167375/Warbringer-new-bb-routed-routing.error-minimal-case.fzz.zip)

(remove the trailing .zip to recover the .fzz file)

In this sketch making a ground connection via the breadboard between the feather and the mpr121 creates an incomplete connection message as in the example sketch above:

[image]

while the connection is correct and generates the rats nest line in schematic, the routing is noted as incomplete at the bottom of the screen. 

**Build:**
Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]
**Operating System:**
Windows 7 pro

**Steps to reproduce:**

- Load the above sketch and "delete wire" on wire521 (the connection from the pr121 part to breadboard ground). The message goes back to "no connections to route". 

- Reconnect a wire in breadboard from the gnd pin of the mpr121 and the breadboard ground rail. The connection creates, but "0 of 1 nets routed - 1 connection still to be routed" appears instead of routing complete. Note however that the feather ground connection is to the ground pin on the part on the breadboard rather than to one of the breadboard pins as it should be (but Fritzing allowed the connection.)

- delete the wire from the feather to ground and replace it with a connection from the breadboard pin to ground and routing completes correctly as shown in the screen shot below so it appears the problem is from the connection from the pin on the part to breadboard that is the issue, since that will likely cause a dual connection to the breadboard one of which won't appear routed.

## Expected Behaviour

The routing complete message as this should appear in both cases, or an error should appear when the connection from the part pin to breadboard is made a second time by connecting to the pin on the part rather than the a breadboard pin of the row that the part is connected to as below:

[image]

KjellMorgenstern, 2020-02-14:
Isn't that the same as #3615 ?

vanepp, 2020-02-14:
Yes now that I look at it more closely, the underlying connection is the same as #3616 in that the connection under the feather on the breadboard isn't connected to anything which makes the underlying connection the same. The issue I saw here is that the connection to the feather is via the pin on the feather (rather than the breadboard) which you couldn't do in real life but can in Fritzing, but with the underlying bug fixed that may not matter as routing should still work. Probably the best bet is to close this one, and try this again once #3616 is fixed and see if there is still a problem (there may not be.)

failiz, 2021-03-02:
Could you try in 0.9.6 to see if this is fixed? #3616 has been fixed and included in that release.

vanepp, 2021-03-11:
Indeed this one appears to be fixed too, so I'll close this. The test case above shows routing complete when loaded and when the wire is added then deleted again.

closed 2021-03-11 by vanepp