fritzing/fritzing-app #3617: Placing a (or any) component fails in breadboard or schmatic area.

Ronkaplan, 2020-02-08:
Os Linux 5.4.17-200.fc31.x86_64 #1 SMP Sat Feb 1 19:00:13 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Fritzing version 0.9.2

vanepp, 2020-02-08:
0.9.2 is now two versions old (0.9.4 is the current Fritzing version.) so support may be limited (I for instance have never run 0.9.2). That said there isn't enough information here to do anything useful. If you are getting error messages about the swapping mechanism not working you may want to try and install  Fritzing-parts via you r app manager. Some of the installable packages did not make the Fritzing-Parts repository a dependency and if it is not there Fritzing will not run (and you will have no parts.) Presumably you installed Fritzing from one of the app managers on whatever linux version this is (i.e. Ubuntu 18.04. fedora, etc,) I would suggest instead installing one of the 0.9.4 tarballs from the Fritzing download site if your linux version is one of those supported. The tarballs for 0.9.3b (which should still be available I think) are self contained just untar the install file and start the application and it should run. The 0.9.4 release requires the listed dependencies on the install page to be installed before Fritzging (mostly Qt libraries that are included in the 0.9.3 tarball.)

KjellMorgenstern, 2020-02-09:
Closing, as it is obviously possible to place parts in breadboard or schematic, in all versions of Fritzing. If you want to re-open the issue, please add details on how to reproduce it on Fritzing version 0.9.4.

closed 2020-02-09 by KjellMorgenstern