fritzing/fritzing-app #3619: Exporting breadboard to svg replaces ic image label with generic

mMerlin, 2020-02-14:
## Current Behaviour

Exporting the breadboard view of sketch to svg image does not match the actual Fritzing Breadboard view very well.  Exporting to a png image handles all of this correctly.  I assume because it is effectively doing a screenshot, where svg needs to actually generate the image from sketch data.

* dip ic label is generic "IC", or blank, instead of the breadboard part image label
* the darker circle at wire bend points is not shown, replaced by the standard wire end border.
* open (not connected) part connector pins are shown in grey instead of red highlight
* wire end points are shown in the standard wire color, without the highlighting circle

[image]
[image]

To be able to upload, renamed to include .zip suffix, but the file is really just an svg
[ic-bb-export.svg.zip](https://github.com/fritzing/fritzing-app/files/4202016/ic-bb-export.svg.zip)
But, here is a screenshot of that file in my local image viewer

[image]

**Build:**

```
          ID: org.fritzing.Fritzing
         Ref: app/org.fritzing.Fritzing/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 0.9.4b
     License: GPL-3.0+
      Origin: flathub
  Collection: org.flathub.Stable
Installation: user
   Installed: 186.2 MB
     Runtime: org.kde.Platform/x86_64/5.13
         Sdk: org.kde.Sdk/x86_64/5.13

      Commit: c595162a1d06f01eab9b66092c7ca68518974b9887295cd59950c439bfba3a72
      Parent: 39bbd0e2b1e2740293803fd4dc550fb0b1827dc5ddccff56a07f7aedad9c2aa5
     Subject: Add OARS info (7dec8699)
        Date: 2019-12-18 19:56:11 +0000
```

**Operating System:**

```
NAME=Fedora
VERSION="31 (Workstation Edition)"
ID=fedora
VERSION_ID=31
VERSION_CODENAME=""
PLATFORM_ID="platform:f31"
PRETTY_NAME="Fedora 31 (Workstation Edition)"
ANSI_COLOR="0;34"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:31"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f31/system-administrators-guide/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=31
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=31
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```

**Steps to reproduce:**

Create a new sketch, go to breadboard view, delete breadboard, place max7219 IC (sparkfun led driver), FRM050 and 4071 from core ICs.  Connect a wire from pin 4 to pin9 (GND to GND) on the MAX729 and create a bend in the wire.

File ¦ Export ¦ as Image ¦ PNG
File ¦ Export ¦ as Image ¦ SVG

Comparing those exported images and The Fritzing screen, shows all of the issues reported.

## Expected Behaviour

The exported svg should have the part label from the breadboard image of the part.  The wire bend rendering is less important, but should match the breadboard view.  The rest is cosmetic.

KjellMorgenstern, 2020-02-16:
The problem, at least for the FRM050, is the font-size:8px , which Fritzing chokes on when exporting.
During export Fritzing parses this into
```
 <g xmlns="http://www.w3.org/2000/svg" stroke="none" fill="#e6e6e6" id="label-1" style="text-align:start;line-height:125%;writing-mode:lr-tb;text-anchor:start;font-family:OCRA" font-size="0">
  <text xmlns="http://www.w3.org/2000/svg" y="10.5702" x="3.74681">FRM</text>
```
To reproduce, set a breakpoint at sketchwidget.cpp:7274  (renderToSvg) and inspect the itemSvg value returned from retrieveSvg

KjellMorgenstern, 2020-02-16:
Actually, the fix only works for the FRM050. The 4071 seems to have a different issue.

mMerlin, 2020-02-16:
And both of those were samples I pulled in to verify that the problem was not unique to the Max7219.  Which is where I spotted it.

commit af82158f (2020-03-16): font-size px is the same as user coordinates and can be ignored. Fixes #3619
closed 2020-03-16 by KjellMorgenstern