fritzing/fritzing-app #3627: Raise and Lower breadboard part incorrect until reopen sketch

mMerlin, 2020-02-25:
## Current Behaviour

With this sketch, moving a part backward on the breadboard view, moved a different part (the one that should have been the back most) to the front instead.  At least that is what it looks like on the screen.  However, an (immediate) svg export showed the requested part move instead.  Save, close, reopen got the breadboard view synchronized with what the previous export showed.

Screen shot and svg (with dummy .zip suffix, just rename to remove) export immediately after opening the sketch.

[image]
[wirewrap-eye1-sequence00a-bb.svg.zip](https://github.com/fritzing/fritzing-app/files/4252621/wirewrap-eye1-sequence00a-bb.svg.zip)

Screen shot and svg (as before) export immediately after send backward

[image]
[wirewrap-eye1-sequence00a2-bb.svg.zip](https://github.com/fritzing/fritzing-app/files/4252628/wirewrap-eye1-sequence00a2-bb.svg.zip)

Screen shot after save, close, reopen

[image]

I have had regular issues manipulating parts using Raise and Lower.  This was a case that I could reproduce, and the research done gathering the details for reporting discovered that it seems to be a display problem.  The SVG export showed what I would expect from the actual operation performed, so I expect the sketch content is correct as well.  But save, close, reopen after every Raise and Lower operation, to see if really got what was intended is not a nice work flow.

**Build:**

Version: 0.9.4b
Commit: c595162a1d06f01eab9b66092c7ca68518974b9887295cd59950c439bfba3a72
with flatpak

**Operating System:**

% cat /etc/os-release
NAME=Fedora
VERSION="31 (Workstation Edition)"
ID=fedora
VERSION_ID=31
VERSION_CODENAME=""
PLATFORM_ID="platform:f31"
PRETTY_NAME="Fedora 31 (Workstation Edition)"
ANSI_COLOR="0;34"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:31"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f31/system-administrators-guide/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=31
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=31
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation

**Steps to reproduce:**

Really .fzz file, just rename to remove the .zip
[wirewrap-eye1-sequence00.fzz.zip](https://github.com/fritzing/fritzing-app/files/4252643/wirewrap-eye1-sequence00.fzz.zip)

* open example file wirewrap-eye1-sequence00 to breadboard view
* take reference screenshot
* do export; as Image; SVG
* select R1 (resistor on the left)
* Part; Raise and Lower; Send Backward
* take screenshot of incorrect screen
* do export; as Image; SVG giving correct view
* save sketch; close sketch; open sketch
* take screenshot of correct screen

## Expected Behaviour

The breadboard view should match the sketch content, which appears to be correct from the svg export.

KjellMorgenstern, 2020-02-26:
Easy to reproduce (0.9.5 develop build)

failiz, 2021-10-02:
It was not caused by a screen refresh issue. The problem was just the code for raising/lowering had a bug. The algorithm assigned the same z value to several parts after the "send to back" command. If the z value is the same, Qt displays the older one on top. I guess that the creation order of the parts changed between the different files. No idea why the svg image showed the change.  

The problem is caused because the algorithm expects that the list of items is ordered by LayerID and Z value. The algorithm sorts the file by Z value but in this case the items are then not sorted by LayerID. This is the list of items as sorted by the code:

```
"Disp1 viewLayerID=2 z=2.50001"
"V6 viewLayerID=9 z=2.50002"
"Ground6 viewLayerID=9 z=2.50003"
"V5 viewLayerID=9 z=2.50006"
"U1 viewLayerID=2 z=2.50009"
"load viewLayerID=9 z=2.5001"
"R2 viewLayerID=2 z=2.50012"
"R1 viewLayerID=2 z=2.50013"
"Wire7823 viewLayerID=3 z=3.5"
"load viewLayerID=9 z=3.50004"
"clk viewLayerID=9 z=3.50005"
"Wire1918 viewLayerID=3 z=3.50007"
"Wire1915 viewLayerID=3 z=3.50008"
(Some wires omitted for the sake of clarity)
"Wire7799 viewLayerID=3 z=3.50145"
"Wire7800 viewLayerID=3 z=3.50146"
"Wire7824 viewLayerID=5 z=6.00147"
"Wire7825 viewLayerID=5 z=6.00148"
"Wire7826 viewLayerID=5 z=6.00149"
```

Notice how the the list is not sorted by viewLayerID. I changed the code to sort by LayerID AND z value. This is the order produced now:

```
"Disp1 viewLayerID=2 z=2.50001"
"U1 viewLayerID=2 z=2.50009"
"R2 viewLayerID=2 z=2.50012"
"R1 viewLayerID=2 z=2.50013"
"Wire7823 viewLayerID=3 z=3.5"
"Wire1918 viewLayerID=3 z=3.50007"
"Wire1915 viewLayerID=3 z=3.50008"
(Some wires omitted for the sake of clarity)
"Wire7824 viewLayerID=5 z=6.00147"
"Wire7825 viewLayerID=5 z=6.00148"
"Wire7826 viewLayerID=5 z=6.00149"
"V6 viewLayerID=9 z=2.50002"
"Ground6 viewLayerID=9 z=2.50003"
"V5 viewLayerID=9 z=2.50006"
"load viewLayerID=9 z=2.5001"
"load viewLayerID=9 z=3.50004"
"clk viewLayerID=9 z=3.50005" 
```
And then the code works as expected.

When new components are introduced, they get a z value of viewLayerID+0.5+ 0.00xx. When the algorithm moves a component (raise/lower/sendtoback/sendtofront), the code changes the z values of to viewLayerID+0.00xx.

Interestinly, in this file, when adding a resistor gets a 3.50147 or so, instead of a 2.5005 (so the wires can be placed under the resistor). But this is another issue and it gets fixed when doing a lowering/raising operation.

failiz, 2021-10-02:
In any case, the code does not take into account where the  items are. If two resistors are far away and they do not overlap, I would say that the lowering/raising the resistor should not consider the other resistor for calculating the new z value.

failiz, 2021-10-15:
I completely agree with what you wrote. I was not suggesting to assign equal z values to two components. I was referring that for me, the raise and lower commands should be local to the stack of objects that are overlapping. And right now, the code uses a global approach.

I completely understand that determining if two component are overlapping is time consuming and complex. However, I feel that the raise and lower commands are not very useful in large files with several components. In that situation, users will need to apply several times the raise/lower command until they are able to see some effect. Of course, bring to bottom and top commands will still work as they are global.

Maybe it is not the optimal solution, but just by considering the bounding boxes of the items, the issue will be reduced. If you talk this into account,  you will only consider a subset of the items, not all the items of the scene as now.

However, this discussion should be another issue if we think that it is important enough. For me, it is ok as it is (with the sorting code fixed) as in most cases you can use several bring to top/bottom commands to set the right order.

In any case, do you remember what you did to get the z order of the items messed? It would be nice to fix that source of this issue.

closed 2022-06-01 by KjellMorgenstern