fritzing/fritzing-app #3628: Parts editor crashes when using keyboard shortcuts to dismiss dialog boxes

chuank, 2020-02-26:
## Current Behaviour
Hitting return on the keyboard to dismiss dialog boxes in the Parts Editor has a high chance of crashing the entire Fritzing application. Using the mouse/trackpad to click on the dialog box buttons works.

**Build:**
Fritzing 0.9.4 CD-498

**Operating System:**
macOS Catalina 10.15.3

**Steps to reproduce:**
Create a new part with Parts Editor. Make some general edits, import SVG etc; anything that will bring up a dialog box. Hit return on the keyboard to dismiss it, leading to crash (not all the time, but chances are quite high)

## Expected Behaviour
Shouldn't crash unexpectedly. All work is lost and not recoverable when this happens.

KjellMorgenstern, 2020-02-26:
Can you reproduce it with the development build? It should give a usable backtrace in case of a crash
https://github.com/fritzing/fritzing-app/releases/download/CD-548/fritzing-3d61c58421bdb63ca903bb5d11310a257f1ec0ed-develop-548.mojave.10.2.dmg

failiz, 2023-02-22:
I could not reproduce this in 0.9.10 (Windows) and the reporter did not answer in 3 years. Close?

closed 2023-02-22 by KjellMorgenstern