fritzing/fritzing-app #3631: can build but not run on Fedora 31

mMerlin, 2020-02-28:
## Current Behaviour

After a build of Fritzing using the steps below, running the `.Fritzing.sh` script as shown on the
[linux notes](https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes) page gives:

```sh
./Fritzing.sh: line 17: /home/phil/development/fritzing-app/lib/Fritzing: No such file or directory
```

A little research shows that the script is setting LD_LIBRARY_PATH to "«repoPath»/lib" and trying to run "«repoPath»/lib/Fritzing".  Neither of which exist.  I can easily run the actual "«repoPath»/Fritzing", but where is the lib folder?  The only lib folder I find in the git repository, is in the src folder.  Looking at a separate flatpak installation on a different machine (0.9.4b), finds the Fritzing executable in files/bin and the library as files/lib (inside $HOME/.local/share/flatpak/app/org.fritzing.Fritzing/x86_64/stable/«commit guid»/)

Side note, the Fritzing executable shows up as an untracked file in the git repository after the build.

**Build:**
error case commit cc8758a1ea47d7d11c3ef5394551c0a3d7f44945 (develop branch)
reference 0.9.4b commit c595162a1d06f01eab9b66092c7ca68518974b9887295cd59950c439bfba3a72

**Operating System:**

NAME=Fedora
VERSION="31 (Workstation Edition)"
ID=fedora
VERSION_ID=31
VERSION_CODENAME=""
PLATFORM_ID="platform:f31"
PRETTY_NAME="Fedora 31 (Workstation Edition)"
ANSI_COLOR="0;34"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:31"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f31/system-administrators-guide/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=31
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=31
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation

**Steps to reproduce:**

spin up virtualbox vm from Fedora-Workstation-Live-x86_64-31-1.9.iso then

```sh
sudo dnf upgrade
reboot
sudo dnf install qt5 qt5-qtserialport-devel libgudev-devel qt5-qtsvg-devel qt-devel qt5-devel libgit2-devel openssl-devel make
mkdir $HOME/development
cd $HOME/development
wget https://dl.bintray.com/boostorg/release/1.72.0/source/boost_1_72_0.tar.gz
sha256sum boost_1_72_0.tar.gz
# c66e88d5786f2ca4dbebb14e06b566fb642a1a6947ad8cc9091f9f445134143f  boost_1_72_0.tar.gz
tar xvfz boost_1_72_0.tar.gz
git clone https://github.com/libgit2/libgit2
cd $HOME/development/libgit2
git co v0.28.4
mkdir build
cd build
cmake -DBUILD_SHARED_LIBS=OFF ..
cmake --build .
cd $HOME/development
git clone https://github.com/fritzing/fritzing-app
cd $HOME/development/fritzing-app
qmake-qt5
sudo make install
./Fritzing.sh
```

## Expected Behaviour

Frittings should run, and report that it needs the parts library.

mMerlin, 2020-03-05:
The problem turned out to be a combination or 2 things.  First I messed up getting the QT sdk bin folder in the path.  Then the information on https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes for running Fritzing is incorrect.  At least for Fedora.  I have updated the information there to what worked, with extra information about working around Wayland, and specifying the parts library path.

closed 2020-03-05 by mMerlin

KjellMorgenstern, 2020-03-05:
I didn't follow where the "sudo make install" got in there, but that will cause problems at least since version 0.9.3b . Did you actually run this last step?

mMerlin, 2020-03-05:
Ran it, ran Fritzing per the adjusted notes. Seemed to work fine with the minimal smoke tests I did.  Running the script specified did not work.  Which is where this issue actually started.  This was done on a VM, and I have rollback snapshots, with notes of what was done, so it will be easy to reproduce if you want some log output.

KjellMorgenstern, 2020-03-05:
The phoenix.pro fle has a unix section that configures various paths, which are then later used by make install. After the install Fritzing will need a part folder location. It looks at some default locations, but when installed to a system dir, it is unlikely to find it by itself. 
The trouble starts when users then copy (as root) the parts folder next to the system install, so that Fritzing can automatically find it, but where it is write protected.

That meas the corrections you did in the wiki are valid, also for other linuxes!

Anyhow, I removed the "sudo make install" step from the instructions. This step is usually not required for development, running Fritzing from the build dir should work as well. And it avoids the write protection problem.

mMerlin, 2020-03-05:
I'll retry my build environment sequence, but from what I saw last time (described in the initial issue post), the Fritzing.sh script expects to find a 'lib' folder, with Fritzing executable inside it.  qmake and make did not seem to create that.  Something is missing from the information.  vanepp mentioned in another thread a release.sh script (in tools) that has additional build logic that might apply.  Especially for building a debug version for gdb.

I did not really want to do an install anyway.  This is supposed to be test/debug/development version, which should run from the repository/build folder, without impacting any version the might also be installed.  For my environment, that distinction does not apply.  The whole VM is the test environment.  The installed version is on the VM host machine.

I you removed the sudo make install, the first part of the adjustment I did for running Fritzing is not going to work.  Fritzing will not be found in the system PATH.  The X11 and parts library path will still work.

KjellMorgenstern, 2020-03-05:
Use qt creator. It will give much better debug info. gdb is not good at expanding qt data types, even looking into a QString requires quite some magic.

Setting up with qt creator I do this:
* get boost, libgit2 and parts folder, all next to the fritzing-app folder.
* build libgit2
* open the phoenix.pro file in qt creator

Not sure if I forgot somehting , maybe you need to set the working directory to your fritzing-app directory in the "Run Settings", but I think you are altready good to go.

mMerlin, 2020-03-05:
I did not know that qt did not play nice with gdb.  So I'll work on getting a dev environment with Qt Creator.  I skimmed over the Qt Creator details in the build process infomation, since I did not think I needed it.

That still does not cover the case where someone just wants to do a local build of the newest (or a specific) release.  Because they want to work with something newer than the package manager (or in my case, flathub/flatpak) provides.  There is something missing in those instructions for a basic command line build and run from the build folder.