fritzing/fritzing-app #3632: Move part to back does nothing, or does the wrong thing

mMerlin, 2020-02-28:
## Current Behaviour

Using Raise and Lower, Move to Back in the breadboard view does nothing at all for one part (not the back one) in this sketch, and messes up the z ordering when used on another part.

Moving U1 to back did not change the z ordering at all.  The raw part geometry was extracted before and after to verify.

Moving R1 to back changed the z geometry value of all parts.  3 were changed to exactly 2.  2 more, include R1 were set to 2.00001, and the final part was set to z=2.00002  Even ignoring the duplicate values, none of those are correct based on the previous z values.  R1 should have ended up the lowest, and J1 should have remained the highest.  If the z value for R1 were the only thing changed, the ordering would have been correct.

Before attempting to move U1 to the back (extra leading white space removed)

```sh
grep -A12 -e "instance module.*path=\"/home" -e "logotext\.fzp" raise-lower-test01.fz | grep -e "geometry z" -e "title"
<title>U1</title>
  <geometry z="2.50007" x="0" y="16.7098">
<title>R2</title>
  <geometry z="2.50003" x="79.7783" y="31.021">
<title>Disp1</title>
  <geometry z="2.50002" x="18" y="0"/>
<title>R1</title>
  <geometry z="2.5001" x="-10.2316" y="31.0752">
<title>J1</title>
  <geometry z="2.50014" x="9.00314" y="0.001458"/>
<title>TXT1</title>
  <geometry z="2.50008" x="0.210197" y="0.274506"/>
```

After attempting to move U1 to the back

```sh
grep -A12 -e "instance module.*path=\"/home" -e "logotext\.fzp" raise-lower-test01.fz | grep -e "geometry z" -e "title"
<title>U1</title>
  <geometry z="2.50007" x="0" y="16.7098">
<title>R2</title>
  <geometry z="2.50003" x="79.7783" y="31.021">
<title>Disp1</title>
  <geometry z="2.50002" x="18" y="0"/>
<title>R1</title>
  <geometry z="2.5001" x="-10.2316" y="31.0752">
<title>J1</title>
  <geometry z="2.50014" x="9.00314" y="0.001458"/>
<title>TXT1</title>
  <geometry z="2.50008" x="0.210197" y="0.274506"/>
```

After moving R1 to the back

```sh
<title>U1</title>
  <geometry z="2" x="0" y="16.7098">
<title>R2</title>
  <geometry z="2.00001" x="79.7783" y="31.021">
<title>Disp1</title>
  <geometry z="2" x="18" y="0"/>
<title>R1</title>
  <geometry z="2.00001" x="-10.2316" y="31.0752">
<title>J1</title>
  <geometry z="2" x="9.00314" y="0.001458"/>
<title>TXT1</title>
  <geometry z="2.00002" x="0.210197" y="0.274506"/>
```

Those z values started as 2.5, 2 zeros, 2 digits.  They ended up as 2.0, 2 zeros, 2 digits, with the final part messed up.  That looks like poor management of the z values.  I have an early sketch snapshot that lead to this version.  There were only 3 parts in that with z values of 2, 2.00001, 2.00002.  It looks like that with the 2.5… values, something may have triggered a resequencing process, which messed up.

I have worked with other code that needed to be able to do key insertions.  Normally those use a larger increment when adding sequential entries, to allow room to insert between without a full resequencing.  The range seen here contains 50014 values.  That is a **LOT** of parts.  Incrementing by 500 during placement, then inserting in the middle of the gap would likely reduce the need to adjust multiple values at once.  Or for single step forward and back, swap the 2 values.  Doing that, and starting in the middle of the available range (2.25000 here) should make it all but impossible to need to cleanup the sequencing.  Swaping 2 adjacent values is always safe  So is taking any one entry, and setting to be one lower that the current minimum, or one higher than the current maximum.  It would take 25000 move to back operations to run out of that space.  Problems could be encountered sooner, if there were options to add the next part before or after a specific part.  Which would then go back to using a larger increment to leave room most of the time.

Not tested, but some symptoms indicate that cloning a part will put it in the z order adjacent to the original part.  Which would then need to have, or create, a gap.

**Build:**
app/org.fritzing.Fritzing/x86_64/stable
Version: 0.9.4b
Commit: c595162a1d06f01eab9b66092c7ca68518974b9887295cd59950c439bfba3a72

**Operating System:**

```sh
% cat /etc/os-release
NAME=Fedora
VERSION="31 (Workstation Edition)"
ID=fedora
VERSION_ID=31
VERSION_CODENAME=""
PLATFORM_ID="platform:f31"
PRETTY_NAME="Fedora 31 (Workstation Edition)"
ANSI_COLOR="0;34"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:31"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f31/system-administrators-guide/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=31
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=31
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```

**Steps to reproduce:**

[raise-lower-test01.fzz.zip](https://github.com/fritzing/fritzing-app/files/4266344/raise-lower-test01.fzz.zip)

* unzip raise-lower-test01.fzz
* `grep -A12 -e "instance module.*path=\"/home" -e "logotext\.fzp" raise-lower-test01.fz | grep -e "geometry z" -e "title"`
* open raise-lower-test01.fz; right click U1; Raise and Lower; Move to back; Save
* `grep -A12 -e "instance module.*path=\"/home" -e "logotext\.fzp" raise-lower-test01.fz | grep -e "geometry z" -e "title"`
* right click R1; Raise and Lower; Move to back; Save
* `grep -A12 -e "instance module.*path=\"/home" -e "logotext\.fzp" raise-lower-test01.fz | grep -e "geometry z" -e "title"`

## Expected Behaviour

The z ordering should change when moving U1 to the back.

When moving R1 to back, either only the z value of R1 should change, or if multiple changed, the relative z order of the other parts should not change, and R1 should end up the lowest.

KjellMorgenstern, 2020-02-28:
Isn't that the same as #3627?

mMerlin, 2020-02-28:
These might in the same block of code, but they are different problems.  #3627 actual does the move properly, but messes up the screen display.  This one does not do the move at all (screen or file) for one case, and moves all the parts in the second case, creating duplicate z values for the other case.  Three different failure cases.  #3627 seems to be in the screen refresh after the move is done.  This looks more like it is in the raise and lower itself.

If I could get the build and run to work, I might be able to trace closer.

KjellMorgenstern, 2020-02-28:
> If I could get the build and run to work, I might be able to trace closer.
I'd recommend using qt creator for this kind of development.

mMerlin, 2020-02-28:
Never used Qt creator.  Does it have built in support for gdb, or do you have to go outside anyway?  Right now, just trying to get an environment setup that can do the build.  And be able to run the result.  Something seems missing in the instructions I have for linux and Fedora.  Which said Qt creator was not needed.

Setting up extras for doing actual development might come later.  First I'd like to be able to create an executable that I can debug / trace with.

mMerlin, 2020-02-29:
Checking other things, I noticed that the grep I was using here missed a part.  A better grep is:

```sh
grep -A12 -e "/fritzing-parts/" -e "logotext\.fzp" -e "/partfactory/" raise-lower-test01.fz | grep -e "geometry z" -e "title"

```

vanepp, 2020-03-03:
@mMerlin "Never used Qt creator. Does it have built in support for gdb, or do you have to go outside anyway? " Yes, Qt creator has built in support for gdb. While it is correct that creator isn't necessary to build Fritzing I at least find it easier to use.

mMerlin, 2020-03-03:
@vanepp Per #3631 I was able to build, but not run, Frtizing on Fedora 31.  At least without Qt creator.  I have not spun up an environment with Qt creator yet, to give that path at try.  If that fails, I can spin up a Ubuntu environment, and try there.

vanepp, 2020-03-04:
Qt creator may not help with the run, what is the error message on run? While I haven't yet tried 0.9.4, on 0.9.3b if you didn't run the release script to move the libraries in to their release position, you need to do a sym link for the libgit2 library in /usr/local/lib for Fritzing to find it. That causes this problem, Fritzing will build but not run because it can't find the libgit2 library (it is around 3 years since I hit this, but I think Qt creator may have run, just trying to run Fritzing stand alone failed so Qt creator may help with that.)

mMerlin, 2020-03-05:
All of the information about what happened is in the linked issue.  I did not need the sym link for the initial attempt, because I cloned the repository, and checked out the version I wanted.  That way, the code was actually in the libgit2 folder.  On my retry, I downloaded the release archive instead, so needed the sym link.  Without that, the build fails, so do not even get a chance to try the run.

The information in the wiki that I was following was incorrect/out of date.  Once I got a version to build and run, I updated the information there.  This did not need to use that release script, but I will likely want to use it later.  Reading through it, I see alternate code path for creating a development/debug version.  Which is what I will want to use with gdb.  But a least now, I have a 0.9.5 build working.

failiz, 2021-10-02:
Duplicate of #3627. See that issue for a full explanation.

KjellMorgenstern, 2021-10-30:
Adding the solved label. Even though it is probably a duplicate, we can just also test this once for the release before closing

failiz, 2022-06-14:
Have you tested in 0.9.10? I think it should be solved (but I did not test it).

commit 287eab6e (2022-06-01): fixes the send to back/front and raise and lower commants (fixes #3632 and #3627)
closed 2023-06-19 by KjellMorgenstern