fritzing/fritzing-app #3633: recent folders as well as files

mMerlin, 2020-03-01:
## Problem

When opening files for a projects, it is sometimes necessary to dig through the folder tree to get to where the files are.  That repeats for each file, even though the previous file was in the same folder.  Recent shows the file, but does not provided access to the folder it is in.

## Proposed Solution

Add folders to the recent list, to allow jumping directly to a folder, then select a file from that folder.

Or change "Recent" to "Recent Files", and add a new "Recent Folders"

KjellMorgenstern, 2020-03-01:
How about starting at the folder of the currently open project by default?

mMerlin, 2020-03-01:
So "Open" would show the contents of the folder of the current (or most recently) opened file?  Instead of showing the recent files list?

That would work, though to prevent 'loss' of the existing recent list being shown, that would need to become a new "Project" entry on the open dialog.  So the user could select to get back to the current functionality.  Or have a configuration default to show either recent files or current project folder (initially) in the open dialog.

failiz, 2023-02-22:
Now "open" shows the most recent folder in 0.9.10 (Windows).  Close or is this just in Windows?

KjellMorgenstern, 2023-02-22:
Same on Linux. Closing.

closed 2023-02-22 by KjellMorgenstern