fritzing/fritzing-app #3636: controlling svg export content from the command line

mMerlin, 2020-03-03:
## Problem

Using the -svg command line option to export all views does not have any control over the content that will be included in the images.  When an export is done from the gui, the content can be adjusted by [de]selecting view content.  Like turning off the ratsnest layer for the breadboard view, or setting the pcb to view from above or below.

## Proposed Solution

Provide options on the command line to do the equivalent control of the export content.  Individual flags might be a bit much to implement.  Possibilities:

Single per view settings parameter that takes comma delimited layer on/off flags
-breadboard "«no»ratsnest,«no»notes,…"

Orientation parameter
-pcbfrom below

Maybe even "-pcbfrom both" (or "all"), to generate 2 pcb exports.

Specify a configuration file, possibly json, possibly config format that can set the view specific options.
-viewsettings «file»

The settings file looks the most generic and flexible, and probably easiest to expand later.  It is harder to *just use* from the command line, but with the advantage that common configurations can be saved and reused.

KjellMorgenstern, 2020-03-05:
What is the use case for this? I like the command line, but how could this issue get enough priority over the others to ever be done?

mMerlin, 2020-03-05:
The case I started from is a project I am working on.  Generating a series of images as part of training/tutorial, but I need to turn off the ratsnest lines before exporting.  With that as a start, I expanded a bit, thinking about what else in the same areas would be useful.  Also for that project, I wanted the view from the bottom of the pcb.  That is somewhat broken (report already), but I have a simple edit I can do on the svg files to do the missing left to right mirroring.

I have needed to regenerate those same image (60ish) of them several times, as I have made adjustments to the sketches.

As far as priority, I don't know.  Some lower priority things get done, because they end up with 'easy start' type issue tags.  Probably does not apply to this one.  But if it is on the list, parts of it might start getting added when something else command line related is being worked on.

KjellMorgenstern, 2020-03-05:
Thanks for the explanation. It is in line with long wished features and important use cases of Fritzing. I hoped to be able to work on something similar for 0.9.5 already (had to drop it, to many other Fritzing construction sites) So, maybe 0.9.6 :-)