fritzing/fritzing-app #3637: Duplicate labels for resistors

KjellMorgenstern, 2020-03-06:
So if I have a standard 1/4watt resistor it becomes "R1", but if I also have a 5W power resistor (which is a different part but also has label R) it will also start as "R1", giving two R1s in the same sketch.

vanepp, 2020-03-07:
At this point, I can't reproduce this (and I am the one that mentioned it as a bug!) The parts where I originally saw it are the AMT103 and YumoE6B2 encoders in core parts, but when I created test parts with the label in both cases changed to enc (from yumo-enc and CUI-enc) the labels correctly increment (which they did not when I originally did the parts, which is why the different labels.) I do see there is an error in that the family name and variant of both parts are identical (which they should not be) which may have caused the original issue (I'm currently testing on 0.9.4 rather than 0.9.3b where the parts were made), but the labels are incrementing correctly now so I expect we can close this issue as not a bug. I'll reopen it if I can reproduce the issue at some point. Sorry for the confusion!

failiz, 2021-07-15:
I tested it with transistors (2 bipolar and one fet) and it works well. Close?

vanepp, 2021-07-20:
Yes I thinks so. I can't reproduce it at this time and can report it again if I can do it again.

failiz, 2021-11-02:
@KjellMorgenstern , you are the only one who can close this issue as you opened it...

closed 2021-11-02 by KjellMorgenstern