fritzing/fritzing-app #3639: Placement is imprecise / rounds input

sfcgeorge, 2020-03-08:
## Current Behaviour

[image]

**Operating System:**

macOS 10.15
Fritzing 0.9.4

**Steps to reproduce:**

- Place a component in PCB view.
- Type exactly where you want it to be in the Placement pane such as 80mm.
- Fritzing changes that value to something else like 79.968mm. 

I have turned off snap to grid. It does the same thing with inches too so I don't think it's a conversion error. It's as if it's rounding but to what and why?

nando2301, 2020-04-14:
The Part Placement is so bad, if you try to move a pice with keyboard this issue is taking place too.

failiz, 2021-11-15:
Still reproducible in 0.9.9. I think it is because fritzing uses internally some DPI

failiz, 2022-05-13:
As @vanepp has figured out in the [forum](https://forum.fritzing.org/t/raspio-triangle-request/15887/20), the numbers are changed based on the zoom´s level of the view. 
I have checked the code and I think I have found the issue. After the the text entry, we call HtmlInfoView::xyEntry(). In this function we call SketchWidget::moveItem(ItemBase * itemBase, double x, double y), which calls SketchWidget::moveByArrow(double dx, double dy, QKeyEvent * event). Nothing wrong until here. The problem is in the SketchWidget::moveByArrow function.

This is the important code in SketchWidget::moveByArrow:

```
QPoint globalPos = mapFromScene(m_mousePressScenePos + QPointF(m_arrowTotalX, m_arrowTotalY));
globalPos = mapToGlobal(globalPos);
moveItems(globalPos, false, rubberBandLegEnabled);
```

The problem is that we are converting the position to an integer (QPoint ). I think that it also converts it to the view coordinate system (in mapFromScene), which makes the rounding dependent on the zoom of the view.

In the moveItems function, the integer (QPoint) is translated into the scene coordinates, which causes the rounding errors. See the code below. 

```
void SketchWidget::moveItems(QPoint globalPos, bool checkAutoScrollFlag, bool rubberBandLegEnabled)
{
	if (checkAutoScrollFlag) {
		bool result = checkAutoscroll(globalPos);
		if (!result) return;
	}

	QPoint q = mapFromGlobal(globalPos);
	QPointF scenePos = mapToScene(q);
```

KjellMorgenstern, 2023-08-13:
I suspected this to have caused #4051, but this is not the case -> Confirmed once more that this only fixes the rounding error.

closed 2023-08-13 by KjellMorgenstern