fritzing/fritzing-app #3640: Schematic view svg export net label rotated incorrectly

mMerlin, 2020-03-09:
## Current Behaviour

In the Schematic view, rotating a net label 180° then flipping it horizontally visually brings it back to the initial view.  However, exporting to svg shows the label text upside down

Screenshot of Fritzing Schematic view
[image]

Exported svg, renamed to append .zip
[flipped_test.svg.zip](https://github.com/fritzing/fritzing-app/files/4308916/flipped_test.svg.zip)

Screenshot of svg in image viewer
[image]

Fritzing sketch, renamed to append .zip
[flipped_test.fzz.zip](https://github.com/fritzing/fritzing-app/files/4308915/flipped_test.fzz.zip)

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]
Fritzing version 0.9.4.2019-12-01.CD-498 - Qt version 5.13.2

**Operating System:**

NAME=Fedora
VERSION="31 (Workstation Edition)"
ID=fedora
VERSION_ID=31
VERSION_CODENAME=""
PLATFORM_ID="platform:f31"
PRETTY_NAME="Fedora 31 (Workstation Edition)"
ANSI_COLOR="0;34"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:31"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f31/system-administrators-guide/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=31
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=31
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation

**Steps to reproduce:**

- In a new sketch, change to the schematic view
- Drag a net label to the view
- Part ¦ Rotate ¦ Rotate 180°
- Part ¦ Flip Horizontally
- File ¦ Export ¦ as Image ¦ SVG…

## Expected Behaviour

The exported image should match the Fritzing Schematic view

KjellMorgenstern, 2020-03-09:
That might be already fixed in 0.9.5dev

mMerlin, 2020-03-09:
Not fixed yet.  I just retested with

Version 0.9.5 
(dCD-548-1-gcc8758a1 2020-02-15) 64 [Qt 5.13.2]

and got the same results.  That is the most recent commit in the develop branch.

KjellMorgenstern, 2022-01-11:
Duplicate of #3186

KjellMorgenstern, 2022-01-11:
Closing in favor of #3186

closed 2022-01-11 by KjellMorgenstern